iDevicesInc/SweetBlue #95: Connecting to device waits until scan interval is complete

ryanidev, 2015-04-06:
When using startScan(Interval...), when connecting to a device, it waits for the interval period before actually connecting to the device. 

If stopScan is called before connecting, this is not an issue, it's only when trying to connect before the interval has elapsed.

ryanidev, 2015-04-06:
Actually, after some testing, this also happens with startPeriodicScan()

dougkoellmer, 2015-04-06:
Further testing shows that essentially using the sample code in the README file, it's the battery read task that is delayed until scan is complete, which currently is a feature in the sense that you don't want lots of reads/writes flooding the job queue and pushing the scan task off indefinitely.

However this situation probably smells like a bug to an outside user and so should be addressed. Implementing #38 should actually make this easy to fix.

c10h22, 2015-12-30:
+1! 
For instance, I am obliged to call stopScan() before writing to a characteristic because it have to be as quick as possible from user side. We are using periodic scan of 5sec with 15sec pause, that means, if user press the button just when scan starts, he have to wait 5seconds before seeing its command executed ;)

commit 6a7c2d0c (2016-01-20): closes #95, will be in next release
commit cf1ce56c (2016-01-20): Merge pull request #156 from iDevicesInc/scanissue

closes #95, reads/writes will now interrupt scanning immediately
closed 2016-01-20 by ryanidev