iDevicesInc/SweetBlue #98: Script in samples repo for creating a new project.

dougkoellmer, 2015-04-20:
Should basically copy/paste hello_ble, find/replace the name in manifest/gradle files, and automatically add it to a config array so that it automatically gets its source copied from AS to eclipse folders like the rest of the projects.

dougkoellmer, 2015-04-20:
Actually probably don't need to add the project to an array in configs.sh...can just do wildcard path matching like in the other scripts to copy/paste the code...same way we copy/paste latest sweetblue JARs into each project without knowing the project names.

dougkoellmer, 2015-04-20:
Need to replace project name in (at least)...
 * Manifest file
 * Gradle file
 * Strings file
 * MyActivity.java file

Might be more.

dougkoellmer, 2015-04-20:
Also have to change path from hello_ble/MyActivity.java to (e.g.) simple_write/MyActivity.java

ryanidev, 2015-04-20:
Added. If no name is provided when calling the script, it will prompt for a name. It's best to keep names without spaces, as I'm sure that would cause issues right now.

closed 2015-04-20 by ryanidev