iDevicesInc/SweetBlue #112: OPERATION_NOT_SUPPORTED when trying to write on WRITE_WITHOUT_RESPONSE characteristic

Adnenovish, 2015-05-26:
Hi,
When trying to write on WRITE_WITHOUT_RESPONSE characteristic, nothing happen (write characteristic request is never called).

In fact, P_Task_Write is not added to queue! The code doing that is never reached as getEarlyOutResult returns new ReadWriteEvent (which is not null) so write_internal exits with earlyOutResult instance with status OPERATION_NOT_SUPPORTED!

Changing this part of getEarlyOutResult in P_ServiceManager resolves the issue (see [862b26b](https://github.com/Adnenovish/SweetBlue/commit/862b26baf5479dfee884f74c956b3e789ca89e55) )

```java
if( (char_native.getProperties() & property) == 0x0 )
		{
			//TODO: Use correct gatt status even though we never reach gatt layer?
			ReadWriteEvent result = new ReadWriteEvent(m_device, uuid, null, type, target, data, Status.OPERATION_NOT_SUPPORTED, gattStatus, 0.0, 0.0);

			return result;
		}
```

to

```java
if( (char_native.getProperties() & property) != 0x0 )
		{
			//TODO: Use correct gatt status even though we never reach gatt layer?
			ReadWriteEvent result = new ReadWriteEvent(m_device, uuid, null, type, target, data, Status.OPERATION_NOT_SUPPORTED, gattStatus, 0.0, 0.0);

			return result;
		}
```

Regards,

dougkoellmer, 2015-05-26:
Thanks for contributing!

The logic in the current codebase using `==` seems correct to me. It basically means that we don't even attempt the operation if the characteristic's declared properties bitmask does not overlap with any "write" bits, i.e.:

```java
BluetoothGattCharacteristic.PROPERTY_WRITE
BluetoothGattCharacteristic.PROPERTY_WRITE_NO_RESPONSE
BluetoothGattCharacteristic.PROPERTY_SIGNED_WRITE
```

By changing to `!=` you're telling SweetBlue to try the operation anyway. Can you confirm that this is working? Can you also tell me the result of `char_native.getProperties()` from your debugger?

Adnenovish, 2015-05-26:
Hi,

char_native.getProperties() gives me 4 while property gives me 0.

[image]

dougkoellmer, 2015-05-26:
Thanks for the information. The problem was actually slightly further upstream in the getProperty() method. Please see #114 for more details.

Thanks again for the report and let me know if that works.

closed 2015-05-26 by dougkoellmer

Adnenovish, 2015-05-26:
it works! thanks a lot!