iDevicesInc/SweetBlue #116: How to write on characteristic of specific service?

Adnenovish, 2015-06-16:
Hi,

I have a device with two main BLE Services: Link Loss & Immediate Alert
For both of these services, I have Alert Level Characteristic: 00002A06

When I call 
```
device.write(UUID.fromString("00002A06-0000-1000-8000-00805f9b34fb"),{0x02}) 
```
it writes ALERT_HIGH (0x02) on Immediate Alert.

How can I write on LinkLoss Service's charachteristic which have the same UUID?

Thanks

dougkoellmer, 2015-06-16:
Hi,

We will add method overloads to cover this case immediately. Thanks for bringing it up and stay tuned in the next few hours.

Adnenovish, 2015-06-16:
Thanks Doug appreciate it!

dougkoellmer, 2015-06-16:
I've added overloads for most read/write-type methods that allows you to pass a service uuid in addition to the characteristic uuid.

Let me know if anything else comes up.

commit 22be892d (2015-06-16): Fixes #116
closed 2015-06-16 by dougkoellmer