iDevicesInc/SweetBlue #117: NullPointerException when trying to write on chracteristic without giving service uuid

Adnenovish, 2015-06-22:
Hi,

I got this exception when using write method without service uuid. Giving service uuid resolves the problem.

java.lang.NullPointerException: Attempt to read from field 'android.os.Handler com.idevicesinc.sweetblue.BleManager.m_mainThreadHandler' on a null object reference
            at com.idevicesinc.sweetblue.BleDevice.write(BleDevice.java:3370)

dougkoellmer, 2015-06-22:
Hi Adnenovish,

This error implies that device.isNull() is true. SweetBlue as a whole tries to never return actual null object references to prevent NullPointerExceptions. As you can see we have failed here and will fix this so it doesn't throw an exception and instead returns an error to your ReadWriteListener, but the call would have failed eventually anyway. In the mean time please check that you are working with a valid BleDevice instance before calling the write method.

Thanks!

Adnenovish, 2015-06-22:
Hi,

Thanks for the answer. I agree with you that it may be caused by null device (not really null as I give BleManager a MAC address that may not be in the vinicity of the smartphone).
Anyway, when I rollback to commit 1c152ab, it works without throughing any exception.
I noticed also some strange behaviour when pulling the last commit. For now, I don't have enough time to pick all the changes that I experienced between commit 1c152ab and the last one. I will certainely dive into it this weekend.
I think this issue can be closed as we decided to work with the last stable commit that works perfectly with our test app.

Again, thanks for your quick answers.

closed 2015-06-22 by Adnenovish