iDevicesInc/SweetBlue #119: FAQ_Technical: Some suggestions

c10h22, 2015-07-15:
Hi,

Just to suggest some additions to FAQ_Technical:

- Meaning of CONNECTING_OVERALL Vs CONNECTING
- Creating a new device from a known MAC address gives this new device to have status DISCOVERED even if it's not in range
- Meaning of ADVERTISING: In some cases of disconnections we observered a transitent status event; entrered: ADVERTISING, DISCONNECTED
then
exited: ADVERTISING

Will try to add some suggestions in next days

Cheers,
Adnene

dougkoellmer, 2015-07-15:
Thanks for the suggestions! Keep them coming. For some of these I may just give more detail in the API docs instead.

c10h22, 2015-07-15:
Doug,

I saw you fixed ADVERTISING state issues in bc82edd.

What about state DISCOVERED when device is created from a simple MAC address?
Maybe default value should be UNDISCOVERED when origin is not scan?

dougkoellmer, 2015-07-16:
The ADVERTISING issue I fixed was when a device is created manually through `BleManager.newDevice()`, connected, then disconnected. SweetBlue could erroneously set the device to ADVERTISING even though it was never found through scanning. So I don't think it matches your original issue and I could not reproduce. If you could provide reproducible code that would be appreciated.

`device.is(DISCONNECTED, true, DISCOVERED, true, ADVERTISING, true)` will return `true` if a device was discovered through scanning (even if it was originally created through `newDevice()`) and `device.is(DISCONNECTED, true, DISCOVERED, true, ADVERTISING, false)` if it was only created through `newDevice()` and never discovered through scanning.

I agree this can be a little confusing but it was designed that way on purpose to match the common use cases I observed while writing BLE apps, and changing it now would be backwards incompatible. I will definitely consider revising this pattern when we make a 2.0 release however.

Thanks again for the feedback.

c10h22, 2015-07-27:
Hi,

Now it works as you described, but...when I use Gatt refresh in BleManagerConfig, first connection after bonding fails every time after service discovering. 
Disabling Gatt refresh, resolves the problem.

Should I open a new issue for this? Did you notice this behaviour from your side?

Regards,
Adnene

dougkoellmer, 2015-07-27:
You should try to avoid `useGattRefresh==true` if possible. I made this a public setting but it causes various problems that SweetBlue cannot account for. The only time where you should try to use it is if your device's service profile changes regularly between connections.

c10h22, 2015-07-27:
Well, I agree with you about the use of refresh() method. 
The problem is that we are facing a big problem with our product which embend Nordic nRF8002. We noticed when looking to btsnoop logs that some times Android does not update connection parameters as expected by nRF8002. 
:( 
(we can't figure out the exact conditions that make this behaviour occurs)

We noticed that 30sec after LE Connection creation, nRF8002 send a Connection Parameter Update request as Android creates by default a connection with ~48ms or 7,5ms as Connection Interval.
Android respond that is Accepted but does not send back a Connection Update Response Completed as expected.
After 30 seconds of the third retry (30sec, 1mn30sec, 2mn30sec), it goes in sleep mode with reason : "Unacceptable Connection Parameter" (3mn after connection) 

[image]

We looked deep into Bluedroid code and found this on line 484 https://goo.gl/xl5b2z:

```c
l2cu_send_peer_ble_par_rsp (p_lcb, L2CAP_CFG_OK, id);
 
                    p_lcb->min_interval = min_interval;
                    p_lcb->max_interval = max_interval;
                    p_lcb->latency = latency;
                    p_lcb->timeout = timeout;
 
                    if (p_lcb->upd_disabled == UPD_ENABLED)
                    {
                        btsnd_hcic_ble_upd_ll_conn_params (p_lcb->handle, min_interval, max_interval,
                                                            latency, timeout, 0, 0);
                        p_lcb->upd_disabled = UPD_UPDATED;
                    }
                    else
                    {
                        L2CAP_TRACE_EVENT0 ("L2CAP - LE - update currently disabled");
                        p_lcb->upd_disabled = UPD_PENDING;
                    }
```
l2cu_send_peer_ble_par_rsp with L2CAP_CFG_OK is what we saw in our btsnoop as Connection Parameter Update Response(Accepted)

[image]

but when p_lcb->upd_disabled is not enabled, Android does not modify his connection interval parameter.

The only place where p_lcb->upd_disabled can be marked as UPD_ENABLED can be found on line 175 (L2CA_EnableUpdateBleConnParams method)
this method is called every time we refresh Gatt cache and while discovering services.
As Services discovery may be done from cache, Android will never reset that variable and so never respond to connection parameter update.

Right now, we are trying to tweak this behaviour to force Android to call L2CA_EnableUpdateBleConnParams by using Gatt Refresh.

dougkoellmer, 2015-07-27:
Ok, my general experience is that Android refuses connection intervals that are too fast, and that unfortunately you have to account for the slowest Android devices on the market. I believe 100 ms is a good lower bound but you will have to test. One thing for sure is that you cannot use the same "fast" connection parameters for Android as you do for iOS. Either you have to choose parameters that work with both, or let your firmware know which device it is connecting to (e.g. by writing to a characteristic) so it can choose the best connection parameters.

c10h22, 2015-07-27:
We use 750ms and 1250ms as connection interval limits to avoid battery leacks for our product.
Android default value is too low, so nRF8002 systematically request connection interval update. We take your advice for the next version V2 of our product, but, we still have to found a solution to force Android to accept connection parameters update or at least to tweak this behaviour as we don't have control of the firmware embedded on nRF8002.

closed 2015-08-28 by c10h22