iDevicesInc/SweetBlue #121: Usage best practices

martinbrylski, 2015-08-21:
To ease the understanding of my questions I want to tell you, how my application looks like:

I want to write a state-machine like process in which my peripheral is configured, sends now and then a working state and unconfigure it after a while.

I currently use multiple BleManager instances, one for each state.

My questions are:
- How can I cancel a `BleManager?` scanning instance, including the deletion (and GC) of the associated `DiscoveryListener`? (to start another BleManager instance - or rather state, and to ensure that only one state is active)
- Do I have to manually disconnect after calling connect() on a device? If yes, where is the best place to do so? After last readWrite? And how do I ensure that the connection was successfully closed and no side effects are still active, like reconnecting?
- As my peripherals alter their name, how can I ensure that the name is updated immediately? For explanation: In Lollipop I see some system-internal `BleScanResult` with the correct advertised name. But the callback delivers only the cached (former) name. As my state-machine also depends on the name I need to have the latest. Is it preferable to use `startPeriodicScan()` over `startScan()`?

--
Besides that I want to thank you for your work regarding a stable ble communication layer.

dougkoellmer, 2015-08-21:
Hi Martin,

 * `BleManager` is actually a singleton, there is no way to create multiple instances. I can explain the reasons for this if you're interested, but `BleManager.get()` is always returning the same instance. I will update API documentation to make this more explicit.
 * The manager can only have one `DiscoveryListener` at a time. `stopScan()` stops the scan started from `startScan()`. Any new listener to `startScan(listener)` releases the reference to the old one. You can call `setListener_Discovery(null)` if you just want to GC it explicitly.
 * Best disconnect behavior depends on your use case, but if you want to explicitly disconnect then use `BleDevice.disconnect()`. No side effects like reconnecting should take place after this. The only other way a device will disconnect is by killing the app, going out of range, bluetooth stack crashing, peripheral being turned off, etc., etc. In these "implicit" cases SweetBlue will try to automatically reconnect by default but this can be controlled through options in `BleDeviceConfig`. See https://github.com/iDevicesInc/SweetBlue/wiki/FAQ_Technical#how-is-reconnection-handled for more information.
 * This might be a bug. I will look into it. Please follow https://github.com/iDevicesInc/SweetBlue/issues/123.
 * Depends on your use case. This is just an explicit way to force SweetBlue to remove all references to the device. To "discover" it again you will either need to scan again or call `BleManager.newDevice()` with the same mac address.
 * You may or may not need to (it's been a while since I tested this with 4.3 more than a year ago). But at the least `pushWakeLock()` can't hurt if you want to be sure.
 * Please see https://github.com/iDevicesInc/SweetBlue/wiki/FAQ_Technical#whats-up-with-bletransaction-when-would-i-need-to-use-this. (I just wrote this)

dougkoellmer, 2015-08-21:
Note that the name not updating should be fixed in 11dd0a5cef45df1c67e745dba576f073fda148d5. I don't have a device on hand that changes its name so please test and confirm the fix. If this *doesn't* fix the issue I'm actually not sure what else can be done.

dougkoellmer, 2015-08-25:
Closing this out as I believe all questions are answered...please create new individual issues if anything else comes up.

closed 2015-08-25 by dougkoellmer