iDevicesInc/SweetBlue #122: ionik Tablet delivers faulty scanRecords

martinbrylski, 2015-08-21:
The [ionik Tablet](http://www.i-onik.de/sites/default/files/TP7_einstieg3.pdf) does not deliver usefull scanRecords.

It is everytime and on all tested peripherals something like: ` 020A000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000`

Where it should be filled with usefull data.

But the name does resolve to a useable string. As data is delivered in one package all data should be recognizable by the bluetooth stack

Is it possible to get the "real" scanrecord?

dougkoellmer, 2015-08-21:
I'm not sure, I've never seen this behavior and we don't have an ionik Tablet to test with. As far as I can see we are always using the same scanRecord provided from the native stack. Are you using `BleDevice.getScanRecord()`?

martinbrylski, 2015-08-24:
Yes, thats what I used. We encountered the same problem on some other low-budget tablets from *Hanspree* and *TerraPad*.

I cant remeber the specific product names. We were not sure if this is a problem on Androids BLE stack or on chipset stack. It is not possible to figure out the specific vendor of the BLE chipset.