iDevicesInc/SweetBlue #123: Peripheral name changes might not be getting respected through scanning.

dougkoellmer, 2015-08-21:
From user question:

As my peripherals alter their name, how can I ensure that the name is updated immediately? For explanation: In Lollipop I see some system-internal BleScanResult with the correct advertised name. But the callback delivers only the cached (former) name. As my state-machine also depends on the name I need to have the latest. Is it preferable to use startPeriodicScan() over startScan()?

martinbrylski, 2015-08-24:
I've found some hints recommending to call [BluetoothDevice.fetchUuidsWithSdp()](http://stackoverflow.com/questions/10793761/how-to-programmatically-clear-the-bluetooth-name-cache-in-android)

It's also not reliable (but there might be some other timing issues since caches gets updated)

dougkoellmer, 2015-08-24:
Did 69bd20e617abccaff81c88f328155a3eba00a2d2 not fix your issue? I'm hesitant to incorporate such thing like fetchUuidsWithSdp() unless absolutely needed, but I will consider it if all else fails.

dougkoellmer, 2015-08-26:
Just bumping this to see if there are any updates here.