iDevicesInc/SweetBlue #124: ReadWriteEvent.onEvent called twice

martinbrylski, 2015-08-21:
I'm writing a characteristic. If it succeeds I write another one. After the second write or if the first write fails I close the connection.

The first write succeeds, but the second write has kind of crazy behaviour: At first it tells it was successful, but little later on it tells it was not succesful because it was cancelled by disconnect.

That's my code:

```
logger.debug("Writing networkid to device {}", myMac);
stateEvent.device().write(MY_UUID, randomNetworkId, new BleDevice.ReadWriteListener() {

    @
    Override
    public void onEvent(ReadWriteEvent netIdWriteEvent) {
        if (netIdWriteEvent.wasSuccess()) {
            logger.debug("Writing network id to device {} was successful. Status {}", myMac, netIdWriteEvent.status().name());

            try {
                bleebDevice.setTested(true);
                databaseHelper.getBleebDeviceDao().createOrUpdate(bleebDevice);

                logger.debug("Updated device {} in database", bleebDevice);
            } catch (SQLException e) {
                logger.error("Could not save device.", e);
            }

            setStateConfiguredAndTesting();
        } else {
            logger.debug("Writing network id to device {} has failed. Status {}", myMac, netIdWriteEvent.status().name());
        }

        stateEvent.device().disconnect();
    }
});
```

and thats the log output:
```
08-21 15:57:13.077  30373-30373/myapp D/myapp.activity.SettingsConfigureFragment﹕ Writing networkid to device 20:73:7a:1a:00:0c
08-21 15:57:13.087  30373-30373/myapp I/P_TaskQueue﹕ BEN(30373) print() - Write(EXECUTING eb_000C 8bc32d3f-5cb9-4d44-bec2-ee689169f62a txn==null) [Write(QUEUED eb_000C 8bc32d3f-5cb9-4d44-bec2-ee689169f629 txn==null)]
08-21 15:57:13.087  30373-30373/myapp I/PA_Task﹕ AMY(30373) setState() - Write(SUCCEEDED eb_000C 8bc32d3f-5cb9-4d44-bec2-ee689169f62a txn==null) - 1290
08-21 15:57:13.087  30373-30373/myapp I/P_TaskQueue﹕ BEN(30373) print() - no current task [Write(QUEUED eb_000C 8bc32d3f-5cb9-4d44-bec2-ee689169f629 txn==null)]
08-21 15:57:13.117  30373-30373/myapp I/P_TaskQueue﹕ BEN(30373) print() - Write(ARMED eb_000C 8bc32d3f-5cb9-4d44-bec2-ee689169f629 txn==null) [queue empty]
08-21 15:57:13.157  30373-30373/myapp D/BluetoothManager﹕ getConnectionState()
08-21 15:57:13.157  30373-30373/myapp D/BluetoothManager﹕ getConnectedDevices
08-21 15:57:13.187  30373-30373/myapp I/P_TaskQueue﹕ BEN(30373) print() - Write(EXECUTING eb_000C 8bc32d3f-5cb9-4d44-bec2-ee689169f629 txn==null) [queue empty]
08-21 15:57:13.207  30373-30389/myapp I/P_BleDevice_Listeners﹕ CAM(30389) onCharacteristicWrite() - char=8bc32d3f-5cb9-4d44-bec2-ee689169f629
08-21 15:57:13.207  30373-30389/myapp I/P_BleDevice_Listeners﹕ CAM(30389) onCharacteristicWrite() - GATT_SUCCESS(0)
08-21 15:57:13.207  30373-30373/myapp D/myapp.activity.SettingsConfigureFragment﹕ Writing network id to device 20:73:7a:1a:00:0c was successful. Status SUCCESS
08-21 15:57:13.267  30373-30373/myapp I/P_TaskQueue﹕ BEN(30373) print() - Write(EXECUTING eb_000C 8bc32d3f-5cb9-4d44-bec2-ee689169f629 txn==null) [Disconnect(QUEUED eb_000C)]
08-21 15:57:13.267  30373-30373/myapp I/PA_Task﹕ AMY(30373) setState() - Write(SOFTLY_CANCELLED eb_000C 8bc32d3f-5cb9-4d44-bec2-ee689169f629 txn==null) - 1292
08-21 15:57:13.267  30373-30373/myapp D/myapp.activity.SettingsConfigureFragment﹕ Writing network id to device 20:73:7a:1a:00:0c has failed. Status CANCELLED_FROM_DISCONNECT
08-21 15:57:13.277  30373-30373/myapp I/P_TaskQueue﹕ BEN(30373) print() - no current task [Disconnect(QUEUED eb_000C)]
```

dougkoellmer, 2015-08-21:
There are some things that are confusing me:
 * I don't see two write calls in your code.
 * I don't see log statements in output resulting from SQL success/failure.
 * Log output implies that SweetBlue is running on another thread, not the main thread. Did you possibly change BleDeviceConfig.runOnMainThread? (If so I strongly suggest not doing so).
 * In short it's hard for me to match your code to the log output.

That said, if you send out a write, then disconnect soon afterwards, the write may "succeed" in the stack underneath, but if disconnect() is called before the write callback, SweetBlue will report that it was cancelled. This was an intentional design decision.

In your case, try waiting for success of the second write before calling disconnect(). Currently your code will always force the device to disconnect after the first write callback if it succeeds *or* fails. See below:

```java
logger.debug("Writing networkid to device {}", myMac);
stateEvent.device().write(MY_UUID, randomNetworkId, new BleDevice.ReadWriteListener() {

    @
    Override
    public void onEvent(ReadWriteEvent netIdWriteEvent) {
        if (netIdWriteEvent.wasSuccess()) {
            logger.debug("Writing network id to device {} was successful. Status {}", myMac, netIdWriteEvent.status().name());

            try {
                bleebDevice.setTested(true);
                databaseHelper.getBleebDeviceDao().createOrUpdate(bleebDevice);

                logger.debug("Updated device {} in database", bleebDevice);
            } catch (SQLException e) {
                logger.error("Could not save device.", e);
            }

            setStateConfiguredAndTesting();

            //--- DRK > Only disconnect if this is the second characteristic.
            if( netIdWriteEvent.charUuid().equals(UUID.fromString("8bc32d3f-5cb9-4d44-bec2-ee689169f629") ) {
                  netIdWriteEvent.device().disconnect();
            }
        } else {
            logger.debug("Writing network id to device {} has failed. Status {}", myMac, netIdWriteEvent.status().name());

           //--- DRK > Disconnect on any kind of failure.
           netIdWriteEvent.device().disconnect();
        }
    }
});
```

martinbrylski, 2015-08-24:
Thanks for your quick answer. I have updated my code snippet to include both writes (and stripped out some uninteresting code). You can see that I already wait for the second write before I disconnect (or if the first one fails).

But you made a good point at checking for the characteristics uuid. I will test that. Are all registered ReadWriteListeners called if any read/write operation has been done (successful or not)? Or only the Listeners registered to an operation on this special uuid?

dougkoellmer, 2015-08-24:
I think I tracked down what could cause this behavior, and I'm surprised it hasn't been discovered yet, so thank you!

Please pull the latest version and confirm that it fixes this issue.

martinbrylski, 2015-08-25:
With the new release it works. Thank you.

closed 2015-08-25 by dougkoellmer

c10h22, 2015-12-30:
Hi all,

I think I have the same behaviour with the last version. I double checked to be sure that my write call is called only once, but when looking to logcat I found two 

```
12-30 02:38:00.210 26802-26802/com.myapp.android I/P_TaskQueue: AMY(26802) print() - no current task [Write(QUEUED device_150D 2f866800-53dc-25b3-0c4a-f0e10c8dee20 txn==null)]
12-30 02:38:00.210 26802-26802/com.myapp.android D/BluetoothManager: getConnectionState()
12-30 02:38:00.210 26802-26802/com.myapp.android D/BluetoothManager: getConnectedDevices

12-30 02:38:00.213 26802-26802/com.myapp.android I/P_TaskQueue: AMY(26802) print() - Write(EXECUTING device_150D 2f866800-53dc-25b3-0c4a-f0e10c8dee20 txn==null) [queue empty]
12-30 02:38:00.218 26802-26802/com.myapp.android I/P_TaskQueue: AMY(26802) print() - Write(EXECUTING device_150D 2f866800-53dc-25b3-0c4a-f0e10c8dee20 txn==null) [queue empty]

12-30 02:38:03.217 26802-26802/com.myapp.android I/P_BleDevice_Listeners: AMY(26802) onCharacteristicWrite_mainThread() - char=2f866800-53dc-25b3-0c4a-f0e10c8dee20
12-30 02:38:03.217 26802-26802/com.myapp.android I/P_BleDevice_Listeners: AMY(26802) onCharacteristicWrite_mainThread() - GATT_SUCCESS(0) 
12-30 02:38:03.240 26802-26802/com.myapp.android I/PA_Task: AMY(26802) setState() - Write(SUCCEEDED device_150D 2f866800-53dc-25b3-0c4a-f0e10c8dee20 txn==null) - 18116
12-30 02:38:03.241 26802-26802/com.myapp.android I/P_TaskQueue: AMY(26802) print() - no current task [queue empty]

```

I think I noticed this issue by looking into sniffed Bluetooth Packets. I can double check this again I put my logs here.

dougkoellmer, 2015-12-30:
@c10h22 That is just some confusing log output. There is only one write there, at least according to the log output, but please confirm using your own logging or breakpoints. There's a task to clean logging up: #92