iDevicesInc/SweetBlue #127: Notification Callback

shashank090789, 2015-08-25:
How to get callback of characteristic change just like BluetoothGattCallback.
I have to  sync 30 days data from pedometer and for each date their is specific write command after 
that pedometer gives the response for each day and onCharacteristicRead is called from BluetoothGattCallback. How would I use this library for the same

dougkoellmer, 2015-08-25:
```java
device.enableNotify(UUID_IN, new ReadWriteListener()
{
    @Override public void onEvent(ReadWriteEvent e)
    {
        if( e.wasSuccess() )
        {
            device.write(UUID_OUT, mData);
        }
    }
}
```

dougkoellmer, 2015-08-25:
Updated the code sample a bit to be more specific.

shashank090789, 2015-08-25:
thanks, it's working now

closed 2015-08-25 by dougkoellmer