iDevicesInc/SweetBlue #131: Sync problem with some devices

shashank090789, 2015-09-04:
Having sync problem with some devices.
one of them is following:
http://www.gsmarena.com/samsung_galaxy_a5-6761.php
not getting any error callback (same as native android BLE)

dougkoellmer, 2015-09-04:
Not sure what you mean by "sync". Please post a code sample demonstrating the issue. SweetBlue should *always* report a callback for all operations, even if it means a TIME_OUT.

shashank090789, 2015-09-05:
following is the code where I ping the device for the data(let me know if I am doing it right),It's working for 
most of devices but I am facing problem with some devices like I mentioned above.

 public void fetchKstepData(final String macAddress)
   {
	
	  if(bleManager.hasDevice(macAddress)&&bleManager.hasDevice(BleDeviceState.DISCOVERED))
	  {
		 bleManager.undiscover(bleManager.getDevice(macAddress));
	  }

		  bleManager.startScan(new BleManager.DiscoveryListener() {

			  @Override
			  public void onEvent(DiscoveryEvent event) {
				  //m_bleManager.stopScan();
				  Log.d("Mac Add", event.device().getMacAddress());
				  //adding device to arraylist

				   availableDevices.add(event.device());
				sendBleBroadcast(ACTION_DEVICE_BLE_STATUS, null, BleStatus.SEARCHING);

				  if (event.device().getMacAddress().equals(macAddress)) {
					  bleManager.stopScan();
					isDeviceSuccessfullyFound=true;
					sendBleBroadcast(ACTION_DEVICE_BLE_STATUS, null, BleStatus.CONNECTING);

					  if (event.was(LifeCycle.DISCOVERED)) {

						  event.device().connect(new BleDevice.StateListener() {
							@Override
							public void onEvent(StateEvent event) {

							  if (event.didEnter(BleDeviceState.INITIALIZED)) {
								sendBleBroadcast(ACTION_DEVICE_BLE_STATUS, null, BleStatus.INITIALIZING);

								event.device().performTransaction(bleTransaction);

							  } else {

							  }
							}
						  }, new BleDevice.ConnectionFailListener() {
							@Override
							public Please onEvent(ConnectionFailEvent connectionFailEvent) {

							  sendBleBroadcast(ACTION_DEVICE_BLE_STATUS,null,getStatus(connectionFailEvent.status()));

							  return null;
							}
						  });
					  }
					else {

					  }

				  }
				else {

				  }
			  }

		  });
	  }

dougkoellmer, 2015-09-09:
Your code looks OK, but I don't understand what you mean by "sync". What are the status codes you are receiving? Is the connection failing? Failing to discover the device? Failing to do reads or writes? Try BleManagerConfig.loggingEnabled=true also and post output of logcat.

shashank090789, 2015-09-15:
I am receiving Status BLE_TURNING_OFF

dougkoellmer, 2015-09-15:
Does this happen consistently? This is a strange error code to receive, because I only ever see it happening when you turn off BLE through the OS Settings, or by using `BleManager.turnOff()`.

dougkoellmer, 2015-09-28:
Closing this out. Let me know if you found a solution on your side.

closed 2015-09-28 by dougkoellmer