iDevicesInc/SweetBlue #141: Build script should use same dependencies on local machine and build server.

dougkoellmer, 2015-10-16:
Currently build is failing on server but succeeding locally due to latest Android SDK being available on local machine but not on server. Three options:

 * Android SDK and all other relevant dependencies are actually in our repository in the scripts folder somewhere and those are used instead of whatever is on local machine (may bloat the repo too much, will have to see how large this stuff is).
 * Android SDK and all other relevant dependencies are pulled remotely when needed (not ideal in case of internet issues, but if it caches then maybe OK).
 * Build can *only* run on the build machine. Local script invocations are forwarded to the server instead (kind of meh about this one, like being able to do local builds)

I'm leaning towards 1 cause of how simple it is, fewer moving parts, but let me know what you think.

ryanidev, 2015-10-19:
I would also prefer 1, however, the sdk itself can chew up a good amount of space. Just the platform sdk folder for 23 takes up over 108MB. This seems like a bit much to put into the repo, what do you think?

ryanidev, 2016-01-26:
Closing this issue. We're using Travis for builds now, so all we have to do is to make sure the travis.yml file is updated, and it will pull down the necessary SDKs.

closed 2016-01-26 by ryanidev