iDevicesInc/SweetBlue #169: BleManager.startScan(Interval.TEN_SECS) makes periodic scan instead of one shot scan

c10h22, 2016-02-01:
Hi there,

I am writing to report weird behaviour when trying to make one shot scan. I tried with a fresh new project and I have the same behaviour.
Here is my BleManagerConfig:

`
        this.enableCrashResolver = true;
        this.loggingEnabled = true;
        this.autoPauseResumeDetection = false;
        this.revertToClassicDiscoveryIfNeeded = false;

        this.autoScanActiveTime = Interval.DISABLED;
        this.autoScanPauseInterval = Interval.DISABLED;
        this.autoScanPauseTimeWhileAppIsBackgrounded = Interval.DISABLED;
        this.autoScanDelayAfterResume = Interval.DISABLED;

        this.reconnectFilter = new DefaultReconnectFilter(
                DefaultReconnectFilter.SHORT_TERM_ATTEMPT_RATE,
                DefaultReconnectFilter.LONG_TERM_ATTEMPT_RATE,
                DefaultReconnectFilter.SHORT_TERM_TIMEOUT,
                DefaultReconnectFilter.LONG_TERM_TIMEOUT
        );

        this.defaultScanFilter = new ScanFilter() {
            @Override
            public Please onEvent(ScanEvent e) {
                List<UUID> advServices = e.advertisedServices();
                return Please.acknowledgeIf(
                        advServices.contains(ServiceUUID.OWNERSHIP.getUUID())
                                || advServices.contains(ServiceUUID.AUTHENTICATION.getUUID())
                );
            }
        };
`

After calling
`
mBleManager.startScan(Interval.TEN_SECS);
`

I got infinite periodic scan :(

`
02-01 23:14:14.277 6551-6551/com.bledevice.android V/BluetoothManager: ⇢ scanForNewbledevice()
02-01 23:14:14.280 6551-6551/com.bledevice.android V/BluetoothManager: ⇢ onEvent(e=StateEvent
02-01 23:14:14.280 6551-6551/com.bledevice.android V/BluetoothManager:    entered = [SCANNING]
02-01 23:14:14.280 6551-6551/com.bledevice.android V/BluetoothManager:    exited  = []
02-01 23:14:14.280 6551-6551/com.bledevice.android V/BluetoothManager:    current = [ON, SCANNING])
02-01 23:14:14.281 6551-6551/com.bledevice.android V/BoundingFragment: ⇢ onEvent(e=StateEvent
02-01 23:14:14.281 6551-6551/com.bledevice.android V/BoundingFragment:    entered = [SCANNING]
02-01 23:14:14.281 6551-6551/com.bledevice.android V/BoundingFragment:    exited  = []
02-01 23:14:14.281 6551-6551/com.bledevice.android V/BoundingFragment:    current = [ON, SCANNING])
02-01 23:14:14.281 6551-6551/com.bledevice.android V/BoundingFragment: ⇠ onEvent [0ms]
02-01 23:14:14.281 6551-6551/com.bledevice.android V/BluetoothManager: ⇠ onEvent [1ms]
02-01 23:14:14.284 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [Scan(QUEUED)]
02-01 23:14:14.284 6551-6551/com.bledevice.android V/BluetoothManager: ⇠ scanForNewbledevice [6ms] = true
02-01 23:14:14.284 6551-6551/com.bledevice.android V/BoundingFragment: ⇠ onCreateView [63ms] = android.widget.LinearLayout{11eb4ba6 V.E..... ......ID 0,0-0,0}
02-01 23:14:14.284 6551-6551/com.bledevice.android V/BoundingFragment: ⇢ onResume()
02-01 23:14:14.285 6551-6551/com.bledevice.android V/BoundingFragment: ⇠ onResume [0ms]
02-01 23:14:14.295 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
02-01 23:14:14.315 6551-6567/com.bledevice.android D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
02-01 23:14:14.319 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
02-01 23:14:19.305 6551-6551/com.bledevice.android I/PA_Task: AMY(6551) setState() - Scan(INTERRUPTED) - 536
02-01 23:14:19.315 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [queue empty]
02-01 23:14:19.315 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [Scan(QUEUED)]
02-01 23:14:19.336 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
02-01 23:14:19.340 6551-6744/com.bledevice.android D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
02-01 23:14:19.346 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
02-01 23:14:24.351 6551-6551/com.bledevice.android I/PA_Task: AMY(6551) setState() - Scan(INTERRUPTED) - 776
02-01 23:14:24.383 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [queue empty]
[... 108 lines cut ...]
02-01 23:15:55.235 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [Scan(QUEUED)]
02-01 23:15:55.255 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
02-01 23:15:55.260 6551-6567/com.bledevice.android D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
02-01 23:15:55.265 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
02-01 23:16:00.269 6551-6551/com.bledevice.android I/PA_Task: AMY(6551) setState() - Scan(INTERRUPTED) - 5351
02-01 23:16:00.281 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [queue empty]
02-01 23:16:00.286 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [Scan(QUEUED)]
02-01 23:16:00.307 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
02-01 23:16:00.313 6551-6567/com.bledevice.android D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
02-01 23:16:00.319 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
02-01 23:16:05.315 6551-6551/com.bledevice.android I/PA_Task: AMY(6551) setState() - Scan(INTERRUPTED) - 5590
02-01 23:16:05.339 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [queue empty]
02-01 23:16:05.341 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [Scan(QUEUED)]
02-01 23:16:05.362 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
02-01 23:16:05.367 6551-6567/com.bledevice.android D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
02-01 23:16:05.374 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
02-01 23:16:10.365 6551-6551/com.bledevice.android I/PA_Task: AMY(6551) setState() - Scan(INTERRUPTED) - 5830
02-01 23:16:10.417 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [queue empty]
02-01 23:16:10.417 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - no current task [Scan(QUEUED)]
02-01 23:16:10.438 6551-6551/com.bledevice.android I/P_TaskQueue: AMY(6551) print() - Scan(EXECUTING) [queue empty]
`

Thanks for your help
Regards,
Adnene

ryanidev, 2016-02-02:
Thanks for pointing this out. Version 2.50.38 which fixes this will be available soon.

closed 2016-02-02 by ryanidev

c10h22, 2016-02-02:
Awesome. Thx

commit 9e9b82fb (2016-02-02): fixes #169