iDevicesInc/SweetBlue #175: Have reconnect logic kick off when calling connect()

ryanidev, 2016-02-16:
When you call connect() on a BleDevice that hasn't connected in the current app session, the Reconnect logic does not happen. This is because there is no native disconnect() called from Android. If you're explicitly calling connect(), I think the reconnect logic should still kick off.

ryanidev, 2016-02-16:
Not doing this. Brain fart, this is what the alwaysUseAutoConnect option is for in BleNodeConfig, and it's extensions.

closed 2016-02-16 by ryanidev

fabiorodella, 2017-02-10:
Hi @ryanidev, I'm just getting started with SweetBlue, and I seem to be facing the same or a similar issue: if I create a device by MAC address and call connect for the first time in this app session, the connection will be retried a few times but eventually it stops trying. Here's the code I'm using to create the connection:

```java
BleDeviceConfig deviceConfig = new BleDeviceConfig();
deviceConfig.forceBondDialog = true;
deviceConfig.alwaysUseAutoConnect = true;

BleDevice bleDevice = bleManager.newDevice(device.getAddress(), deviceConfig);
bleDevice.connect(e -> {
   ...
```

And here are the relevant logs:
```
02-10 15:50:20.987 22121-22121/ D/BluetoothSweetBlueService: New state for A&D_UC-352BLE_459997: StateEvent
                                                                                                        device     = a&d_uc_9997
                                                                                                        entered    = [CONNECTING_OVERALL]
                                                                                                        exited     = [DISCONNECTED]
                                                                                                        current    = [DISCOVERED, BONDED, CONNECTING_OVERALL]
                                                                                                        gattStatus = GATT_STATUS_NOT_APPLICABLE(0)
02-10 15:50:20.989 22121-22121/ D/BluetoothSweetBlueService: New state for A&D_UC-352BLE_459997: StateEvent
                                                                                                        device     = a&d_uc_9997
                                                                                                        entered    = [CONNECTING]
                                                                                                        exited     = []
                                                                                                        current    = [DISCOVERED, BONDED, CONNECTING_OVERALL, CONNECTING]
                                                                                                        gattStatus = GATT_STATUS_NOT_APPLICABLE(0)
02-10 15:50:21.613 22121-22121/ D/BluetoothManager: getConnectionState()
02-10 15:50:21.613 22121-22121/ D/BluetoothManager: getConnectedDevices
02-10 15:50:21.615 22121-22121/ D/BluetoothManager: getConnectionState()
02-10 15:50:21.615 22121-22121/ D/BluetoothManager: getConnectedDevices
02-10 15:50:21.616 22121-22121/ D/BluetoothManager: getConnectionState()
02-10 15:50:21.616 22121-22121/ D/BluetoothManager: getConnectedDevices
02-10 15:50:21.618 22121-22121/ D/BluetoothGatt: connect() - device: 6C:EC:EB:45:99:97, auto: false
02-10 15:50:21.618 22121-22121/ D/BluetoothGatt: registerApp()
02-10 15:50:21.618 22121-22121/ D/BluetoothGatt: registerApp() - UUID=5cffb800-212f-45a3-a61f-557535ce2b6c
02-10 15:50:21.620 22121-22132/ D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
02-10 15:50:34.114 22121-22121/ D/BluetoothGatt: close()
02-10 15:50:34.114 22121-22121/ D/BluetoothGatt: unregisterApp() - mClientIf=5
02-10 15:50:34.126 22121-22121/ D/BluetoothSweetBlueService: New state for A&D_UC-352BLE_459997: StateEvent
                                                                                                        device     = a&d_uc_9997
                                                                                                        entered    = [DISCONNECTED]
                                                                                                        exited     = [BONDED, CONNECTING_OVERALL, CONNECTING]
                                                                                                        current    = [DISCOVERED, DISCONNECTED]
                                                                                                        gattStatus = GATT_STATUS_NOT_APPLICABLE(-1)
02-10 15:50:34.132 22121-22121/ D/BluetoothSweetBlueService: New state for A&D_UC-352BLE_459997: StateEvent
                                                                                                        device     = a&d_uc_9997
                                                                                                        entered    = [CONNECTING_OVERALL]
                                                                                                        exited     = [DISCONNECTED]
                                                                                                        current    = [DISCOVERED, CONNECTING_OVERALL]
                                                                                                        gattStatus = GATT_STATUS_NOT_APPLICABLE(0)
02-10 15:50:34.135 22121-22121/ D/BluetoothSweetBlueService: New state for A&D_UC-352BLE_459997: StateEvent
                                                                                                        device     = a&d_uc_9997
                                                                                                        entered    = [CONNECTING]
                                                                                                        exited     = []
                                                                                                        current    = [DISCOVERED, CONNECTING_OVERALL, CONNECTING]
[... cut ...]
                                                                                                        gattStatus = GATT_STATUS_NOT_APPLICABLE(-1)
```

Also note that even though I'm setting `alwaysUseAutoConnect` to `true` in the device config, it seems to be using `false` in the connectGatt call:
```
BluetoothGatt: connect() - device: 6C:EC:EB:45:99:97, auto: *false*
```

How can I make sure that SweetBlue will _always_ keep trying to connect to the device until I call `disconnect()` myself?

fabiorodella, 2017-02-13:
After closer investigation, it definitely seems that my problems are caused by the `alwaysUseAutoConnect` flag not being used on Marshmallow and up. I created a separate issue for this: https://github.com/iDevicesInc/SweetBlue/issues/238