iDevicesInc/SweetBlue #211: Restarting scan

kentcb, 2016-07-21:
Hi,

I'm trying to restart a scan but it doesn't seem to be working. The initial scan works, but a subsequent one does not.

The basic flow is:
* call `BluetoothEnabler.Start(activity, enablerFilter)`
* in `EnablerFilter.onEvent`, check `IsDone` and if so call `BleManager.StopAllScanning` then `BleManager.StartScan(scanFilter, discoveryListener)`
* in `ScanFilter.onEvent`, acknowledge where relevant
* in `DiscoveryListener.onEvent`, check if the event is `Discovered` and, if so, utilize the device within

In debugging my app, I can see that the first scan works perfectly. The second one successfully calls `StopAllScanning` and `StartScan`, but my `ScanFilter.onEvent` is never called.

Am I doing something wrong?

ryanidev, 2016-07-21:
What you're doing looks right (as long as you are passing the scanfilter in that second startScan call, or alternatively, setting it in BleManagerConfig). Are you using the latest version? (2.51.48)

kentcb, 2016-07-22:
My bad. There was an exception being raised (and swallowed) in a reactive pipeline that was the root cause of this problem.

kentcb, 2016-07-22:
I take that back. There was another issue but it turns out that wasn't affecting the scan like I thought. So I'm back to square one, with a scan that works once only.

The version I'm using is 2.51.48, running on Nexus 5. Here is logcat output from my running the app, performing a successful scan, then performing a second scan that returns nothing: https://gist.github.com/kentcb/12908c10f8b2eafa62d5186c283709d6

Any ideas what's going on here?

ryanidev, 2016-07-22:
Not sure what's going on there. Looks like you're using Xamarin, which we don't support yet. For debugging purposes, turn on logging in SweetBlue to see what it's doing.

BleManagerConfig.loggingEnabled = true; // It's false by default

kentcb, 2016-07-25:
Thanks @ryanidev. Yeah, I'm painfully aware you guys don't support Xamarin (what's up with that?) because I had to create bindings myself :(

Hmm, I thought it would be straightforward to enable logging. However, since I'm using `BluetoothEnabler` (which does not appear to take a `BleManagerConfig` instance), I assume I have to configure the manager prior to using `BluetoothEnabler`?

Here's what I've done (before using `BluetoothEnabler`):

```
var m = BleManager.Get(Application.Context);
var config = new BleManagerConfig();
config.LoggingEnabled = true;
m.SetConfig(config);

m.UhOh += (s,e) =>
{
    logger.Warn("Uh-oh occurred: {0}", e.P0);
};
```

Does this seem reasonable to you? I don't really know what I'm looking for in [the output](https://gist.github.com/kentcb/12908c10f8b2eafa62d5186c283709d6), but I don't see anything different. Also, there is no uh-oh raised.

kentcb, 2016-07-25:
I have reproduced this using only Java code. You can find the repro [here](https://github.com/kentcb/SweetBlueReScanRepro).

ryanidev, 2016-07-25:
Ahh, I think I see your issue now. A device will only get DISCOVERED once. Once the device is discovered, it will then show up in the DiscoverListener as REDISCOVERED.

kentcb, 2016-07-25:
@ryanidev actually, I had already tried that in my C# code but forgot to include it in my Java repro. If this is what you meant, it still doesn't work:

```java
if( e.was(LifeCycle.DISCOVERED) || e.was(LifeCycle.REDISCOVERED) )
```

I've updated my repro accordingly. Can you please try running it and let me know what the problem is?

Thanks

ryanidev, 2016-07-26:
I tried the repo, and it worked for me.

A couple things after looking at the repo a little more closely.

1. The BluetoothEnabler should be run when the app first opens, it only needs to be run once (not every time you start scanning). Though I suppose you could make the argument that the user could shut off bluetooth/remove permissions while your app is running.

2. On Android, you can only do one thing at a time with Bluetooth. So, if you are connecting to a device, you will not see scan results.

3. Connecting to every device you see is a good way to destroy the bluetooth stack. Don't do this unless you absolutely have to. Just because you discover a device doesn't mean you have to connect to it. This is most likely the reason you don't see scan results, as it's trying to connect to every device it sees (and perhaps you're already connected to those devices, so they won't be advertising).

closed 2016-07-26 by ryanidev

kentcb, 2016-07-26:
Thanks for taking a look @ryanidev. I'm not sure why you're raising points 2 and 3 because I'm not doing those things in the repro. It's the repro that isn't working me for, so if it is for you then...that's weird. Maybe something to do with my device?

Also, regarding point 1, why do you say "should be run when the app first opens"? Is there a problem if it gets run every time? I assume not, since the repro works for you.

What do I need to do to get this re-opened? It's a little frustrating for it to be closed when the exact same repro doesn't work for me.

ryanidev, 2016-07-26:
I bring up 2 and 3, because you're connecting to every device that gets discovered/rediscovered. From the repo:

if( e.was(LifeCycle.DISCOVERED) || e.was(LifeCycle.REDISCOVERED) )
                {
                    e.device().connect(new BleDevice.StateListener()
                    ....

That e.device().connect method connects to the device. So, if you're connected to it, you will not see it in another scan, because the device will no longer be advertising. Remove the connect method, and try it again.

kentcb, 2016-07-26:
omg, apologies @ryanidev. I was getting confused between the state of the repro and state of my actual code. Also, I shouldn't rush in replies just before I go to bed :)

You're right of course - the repro was just copy/pasted from the readme and was connecting to every device it encountered. My real code is not doing this, but my real code did not include the _rediscovery_ test. When I added it originally I found that devices were coming through many times and I thought _hmm, weird, maybe I shouldn't be using that_.

I've added it back in and am now storing discovered devices by name in a hash set. Only unique device names result in propagation of an application event. Does this sound like the normal approach to you? Is there perhaps another flag I can set to only receive unique re-discovery events?

Thanks again.