iDevicesInc/SweetBlue #214: Crash

stefreak, 2016-08-05:
```
08-05 14:23:59.978 7853-8092/eu.meshine.meshineng E/AndroidRuntime: FATAL EXCEPTION: Thread-14063
                                                                    Process: eu.meshine.meshineng, PID: 7853
                                                                    Theme: themes:{default=overlay:system, iconPack:system, fontPkg:system, com.android.systemui=overlay:system, com.android.systemui.navbar=overlay:system}
                                                                    java.util.ConcurrentModificationException
                                                                        at java.util.HashMap$HashIterator.nextEntry(HashMap.java:787)
                                                                        at java.util.HashMap$KeyIterator.next(HashMap.java:814)
                                                                        at com.idevicesinc.sweetblue.P_BluetoothCrashResolver.saveState(P_BluetoothCrashResolver.java:380)
                                                                        at com.idevicesinc.sweetblue.P_BluetoothCrashResolver.processStateChange(P_BluetoothCrashResolver.java:271)
                                                                        at com.idevicesinc.sweetblue.P_BluetoothCrashResolver.forceFlush(P_BluetoothCrashResolver.java:250)
                                                                        at com.idevicesinc.sweetblue.P_Task_CrashResolver$1.run(P_Task_CrashResolver.java:43)
                                                                        at android.os.Handler.handleCallback(Handler.java:739)
                                                                        at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                        at android.os.Looper.loop(Looper.java:148)
                                                                        at com.idevicesinc.sweetblue.P_TaskQueue$1.run(P_TaskQueue.java:62)
```

ryanidev, 2016-08-31:
Could you provide some more context on this? What were you doing when this happened?

Could you perhaps provide a way to reproduce the crash?

Thanks

stefreak, 2016-09-05:
Sorry I don't remember anymore.

closed 2016-09-05 by stefreak