iDevicesInc/SweetBlue #216: Disabling auto reconnection

kentcb, 2016-08-31:
If I connect to a device, power it off, then turn it back on again, the device is automatically re-connected. Sounds like a good thing, but in the case of the particular app I'm writing, the client does not want this behavior.

I thought it would be pretty simple to fix: pass in a `ConnectionFailListener` that returns `DoNotRetry` from `OnEvent`. However, this is not working. My connection fail listener does not even get called.

How can I prevent auto-connection from occurring after a connection fails?

ryanidev, 2016-08-31:
Create a ReconnectFilter, and set it in your BleManagerConfig. Just instantiate, and return Please.stopRetrying().

kentcb, 2016-08-31:
Thanks @ryanidev. That does indeed appear to work. I'm still unsure why my original approach would not work, or what the `ConnectionFailListener` is supposed to do, though.

closed 2016-08-31 by kentcb

idevbuilder, 2016-09-01:
The reconnect logic was added in an effort to make sure you can actually
connect to a device. Sometimes, a connect will just fail for no good
reason, and you simply have to retry. The ConnectionFailListener reports
after the retry logic has run it's course.