iDevicesInc/SweetBlue #228: Memory leak in BluetoothEnabler

generic-usver, 2017-01-04:
A lot of tutorials, even the official ones, encourage simply calling `BluetoothEnabler.start(this);` in the Activity for permissions.

This triggers Activity instance assigning to **BluetoothEnabler** class variable, which isn't cleaned up and prevents the Activity from being garbage collected and creates a memory leak:

```
private BluetoothEnabler(Activity activity, BluetoothEnablerFilter enablerFilter)
    {
        m_defaultActivity = activity; // this one
```

It should be either made as a WeakReference to allow auto cleanup or provide some other way of cleanup.

Steps to reproduce:
 1) Attach some kind of leak detector (https://github.com/square/leakcanary is a great and simple one)
 2) Open a new Activity, where the `BluetoothEnabler.start(this)` is called.
 3) Close the Activity
 4) Watch the leak detector complain about **static BluetoothEnabler.s_instance->m_defaultActivity**

Workarounds tried:
 mEnabler = BluetoothEnabler(this);
 Activity @Override onPause() -> calling mEnabler = null; // does not work

ryanidev, 2017-01-04:
Thanks for the report. We're in the middle of refactoring some things, the BluetoothEnabler class among them. This will be fixed in the next feature release (when we go to 2.52.X).

ryanidev, 2017-04-04:
Fixed as of 2.52.01

closed 2017-04-04 by ryanidev