iDevicesInc/SweetBlue #233: Multiple devices issue

flavbaum, 2017-01-13:
Hello, 
I'm trying to implement multiple device connection, for example:
1. Start Scan (normal one)
2. Discover first device, connect, and start read/write operations
3. Discover second device, connect, and start read/write operations
4. Stop scan since no more need.

It appears:
1. While scan is running, read/write operations are slow, like interrupted during 5-6 seconds.
2. If a device is in reconnecting state (being off), scan result won't be transfered during ~19 seconds, or sooner if I disconnect the device (cancel connection).

Sound like it's a question of tasker activity. Any settings / mode I should override to improve this behavior?

Thanks.

ryanidev, 2017-01-13:
Try this. Discover all the devices you want to connect to first. Stop the scan, then connect and do what you need to do on each device.

flavbaum, 2017-01-13:
Thank you for your reply.
I can't do it in a sequential way, but have no worries interrupting the scan on device connection, and automatically restart it after the first characteristics are read/written or based on a timeout. 
So if I read your answer correctly, it appears this is an intrinsic behavior of Sweetblue, right?

ryanidev, 2017-01-13:
The problem is that on Android is that you can only do one bluetooth thing at a time. Scanning is the most intensive task you can do. 

If you're read/writing characteristics every time you connect, consider using an Init Transaction, and pass that into the connect method. Then, once the device is in the INITIALIZED state, you know your reads and writes have completed.

flavbaum, 2017-01-13:
Makes sense, will try so. 
When reconnecting to a device, is a scan started automatically by the lib? 
Because even if I don't start the scan manually, having a (re)connection to a non-existing device (which is off actually) seems blocking the initialization of another one (connect, discover services, read/write some chars.).
BTW I tried contacting your company for commercial question on support@idevicesinc.com with no success, can you help?

ryanidev, 2017-01-13:
You don't need to scan to connect to a device. You only need to scan to discover it. If you're trying to connect to one device, nothing else will happen until the first device either connects, or the connect times out.

The support email is more for our company as a whole (we do a more than just SweetBlue). For Sweetblue specific inquiries, use sweetblue@idevicesinc.com.

flavbaum, 2017-01-13:
Thank you for the confirmation. 
Indeed I copied sweetblue@idevicesinc.com too. Just resent right now, you should have received my request.

ryanidev, 2017-06-28:
Handled outside of github

closed 2017-06-28 by ryanidev