iDevicesInc/SweetBlue #282: startPeriodicScan error in the logs (not crashing)

dneykov, 2017-05-23:
Hello,
I just update to latest version of SweetBlue (great library thanks to all contributors). But when I start my app I see an error each before startScan. The device which I'm testing is Lenovo s60. Here is my log

```
05-23 09:53:19.991 27940-27940/app.identifier D/BluetoothAdapter: startLeScan(): null
05-23 09:53:20.040 27940-27958/app.identifier D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
05-23 09:53:21.521 27940-27940/app.identifier I/PA_Task: MAIN setState() - Scan(TIMED_OUT 488702059 ) - 320
05-23 09:53:21.521 27940-27940/app.identifier D/BluetoothAdapter: stopLeScan()
05-23 09:53:22.027 27940-27940/app.identifier E/ScanTask: Task created
05-23 09:53:22.027 27940-27940/app.identifier W/System.err: java.lang.Exception
05-23 09:53:22.028 27940-27940/app.identifier W/System.err:     at com.idevicesinc.sweetblue.P_Task_Scan.<init>(P_Task_Scan.java:27)
05-23 09:53:22.028 27940-27940/app.identifier W/System.err:     at com.idevicesinc.sweetblue.BleManager.startScan_private(BleManager.java:1697)
05-23 09:53:22.028 27940-27940/app.identifier W/System.err:     at com.idevicesinc.sweetblue.P_ScanManager.update(P_ScanManager.java:303)
05-23 09:53:22.028 27940-27940/app.identifier W/System.err:     at com.idevicesinc.sweetblue.BleManager.update(BleManager.java:3217)
05-23 09:53:22.028 27940-27940/app.identifier W/System.err:     at com.idevicesinc.sweetblue.BleManager$UpdateRunnable.run(BleManager.java:3380)
05-23 09:53:22.028 27940-27940/app.identifier W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
05-23 09:53:22.029 27940-27940/app.identifier W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
05-23 09:53:22.029 27940-27940/app.identifier W/System.err:     at android.os.Looper.loop(Looper.java:135)
05-23 09:53:22.029 27940-27940/app.identifier W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:5261)
05-23 09:53:22.029 27940-27940/app.identifier W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
05-23 09:53:22.029 27940-27940/app.identifier W/System.err:     at java.lang.reflect.Method.invoke(Method.java:372)
05-23 09:53:22.029 27940-27940/app.identifier W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:939)
05-23 09:53:22.029 27940-27940/app.identifier W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:734)
05-23 09:53:22.549 27940-27940/app.identifier D/BluetoothAdapter: startLeScan(): null
05-23 09:53:22.557 27940-27958/app.identifier D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
05-23 09:53:24.062 27940-27940/app.identifier I/PA_Task: MAIN setState() - Scan(TIMED_OUT 979083361 ) - 432
05-23 09:53:24.062 27940-27940/app.identifier D/BluetoothAdapter: stopLeScan()
05-23 09:53:24.573 27940-27940/app.identifier E/ScanTask: Task created
05-23 09:53:24.573 27940-27940/app.identifier W/System.err: java.lang.Exception
05-23 09:53:24.573 27940-27940/app.identifier W/System.err:     at com.idevicesinc.sweetblue.P_Task_Scan.<init>(P_Task_Scan.java:27)
05-23 09:53:24.573 27940-27940/app.identifier W/System.err:     at com.idevicesinc.sweetblue.BleManager.startScan_private(BleManager.java:1697)
05-23 09:53:24.573 27940-27940/app.identifier W/System.err:     at com.idevicesinc.sweetblue.P_ScanManager.update(P_ScanManager.java:303)
05-23 09:53:24.573 27940-27940/app.identifier W/System.err:     at com.idevicesinc.sweetblue.BleManager.update(BleManager.java:3217)
05-23 09:53:24.574 27940-27940/app.identifier W/System.err:     at com.idevicesinc.sweetblue.BleManager$UpdateRunnable.run(BleManager.java:3380)
05-23 09:53:24.574 27940-27940/app.identifier W/System.err:     at android.os.Handler.handleCallback(Handler.java:739)
05-23 09:53:24.574 27940-27940/app.identifier W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:95)
05-23 09:53:24.574 27940-27940/app.identifier W/System.err:     at android.os.Looper.loop(Looper.java:135)
05-23 09:53:24.574 27940-27940/app.identifier W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:5261)
05-23 09:53:24.574 27940-27940/app.identifier W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
05-23 09:53:24.574 27940-27940/app.identifier W/System.err:     at java.lang.reflect.Method.invoke(Method.java:372)
05-23 09:53:24.574 27940-27940/app.identifier W/System.err:     at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:939)
05-23 09:53:24.574 27940-27940/app.identifier W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:734)
05-23 09:53:25.088 27940-27940/app.identifier D/BluetoothAdapter: startLeScan(): null
```

Any suggestions will be appreciated.
Thanks.

dneykov, 2017-05-23:
To me it looks like it is just a debugging message here: https://github.com/iDevicesInc/SweetBlue/blob/6e814f5c654c7571c957275d07c0ca8cd2284be0/app/src/main/java/com/idevicesinc/sweetblue/P_Task_Scan.java#L26-L27

ryanidev, 2017-05-24:
Fixed in 2.52.07

closed 2017-05-24 by ryanidev