iDevicesInc/SweetBlue #301: Incorrect write with bytes amount bigger than MTU

AndZp, 2017-06-20:
Issue: 
When I try to write characteristic byte array with length bigger than MTU I lost 3 bytes  in  the end ofevery stripe.

In the library I've found class : **P_StripedWriteTransaction** : 
I think the method  **void start(BleDevice device)**  works not correctly: 
In line: 
` int end = Math.min(allData.length, curIndex + device.getMtu()) ` - you have to use Effective Write MTU size (which is  already in your code : `getMtu() - BleManagerConfig.GATT_WRITE_MTU_OVERHEAD)`  instead of device.getMtu()

ryanidev, 2017-06-20:
Yes, you're right, thank you. It will be fixed in the next release.

ryanidev, 2017-06-28:
Fixed in 2.52.09 (to be released soon)

closed 2017-06-28 by ryanidev