iDevicesInc/SweetBlue #336: Duplicate characteristic UUIDs

Dave-Osborne, 2017-07-07:
Hi,

We have a peripheral which has a notify characteristic and a separate write characteristic (within the same service) which happen to share the same UUID.

When trying the Write to the UUID the message 'OPERATION_NOT_SUPPORTED' is returned, I suspect it's picking up the Notify characteristic instead.

In an ideal world I'd have the peripheral changed to use unique UUIDs, however this isn't possible.

I have read elsewhere that the BLE specification does allow duplicate UUIDs, would you be able to change your lib. to support this?

The lib. we are currently using supports the duplicate UUIDs but suffers the usual BLE connection issues which is why I would like to trial Sweetblue as a replacement lib.

ryanidev, 2017-07-07:
It's possible to do this now. Per the Bluetooth spec, if you have 2 characteristics in the same service with the same UUID, then each of those should have a descriptor with different values, which is how you know which one to read/write.

So, with SweetBlue, you will need to use a DescriptorFilter to determine which one to perform the operation on.

I suggest using the WriteBuilder class to build your write.
```
WriteBuilder write = new WriteBuilder(myServiceUuid, myCharUuid)
                .setDescriptorFilter(new DescriptorFilter()
                {
                    @Override
                    public Please onEvent(DescriptorEvent event)
                    {
                        byte[] descValue = event.value();
                        // Do check against descriptor value here, then return appropriate Please
                        return Please.acceptIf(myCondition);
                    }

                    @Override
                    public UUID descriptorUuid()
                    {
                        return myDescriptorUuid;
                    }
                }).setBytes(myByteArray);
myBleDevice.write(write, new BleDevice.ReadWriteListener()
        {
            @Override
            public void onEvent(ReadWriteEvent e)
            {
                if (e.wasSuccess())
                {
                    ...
                }
                else
                {
                    ...
                }
            }
        });
```

Dave-Osborne, 2017-07-10:
Thanks Ryan,

Our notify char. has a descriptor, but the write doesn't, so this option doesn't seem to work for me.

ryanidev, 2017-07-12:
Ok, I'll add something quick and dirty for the next release (2.52.10). Basically, you still use the above logic, but return null for the descriptorUuid. Then the onEvent method will fire for all relevant characteristics with the UUID you provide, and you can then determine if you're looking at the correct characteristic or not (maybe by checking if the char has the notification desc, if it doesnt, then that's probably the one you want).

Dave-Osborne, 2017-07-13:
Brilliant, thanks Ryan!

ryanidev, 2017-08-09:
Fixed in 2.52.10 release.

closed 2017-08-09 by ryanidev