iDevicesInc/SweetBlue #345: Potential IndexOutOfBoundsException when calling BleManager.stopScan

iancanderson, 2017-07-21:
We get these every once in a while in our app. Maybe there is a race condition in here somewhere?

I think this is the line that is failing, since ArrayList#remove can throw if the index is out of bounds:

```java
PA_Task task = m_queue.remove(index);
```

Stacktrace:
```
ArrayList.java:255 java.util.ArrayList.throwIndexOutOfBoundsException	
ArrayList.java:308 java.util.ArrayList.get	
P_TaskQueue.java:571 com.idevicesinc.sweetblue.P_TaskQueue.clearQueueOf	
BleManager.java:2226 com.idevicesinc.sweetblue.BleManager.stopScan_private	
BleManager.java:2202 com.idevicesinc.sweetblue.BleManager.stopScan	
```

ryanidev, 2017-07-26:
Thanks for the report. We will have something in the next release to prevent this from happening.

ryanidev, 2017-08-09:
Fixed in 2.52.10

closed 2017-08-09 by ryanidev