iDevicesInc/SweetBlue #349: Manager crashes randomly with nullpointerexpection on getState

kinber95, 2017-08-01:
Hi I get following error sometimes when my app crashes.
`08-01 11:20:09.512 9687-18200/de.app.haveltec.ilockit E/AndroidRuntime: FATAL EXCEPTION: Thread-16
                                                                        Process: de.app.haveltec.ilockit, PID: 9687
                                                                        java.lang.NullPointerException: Attempt to invoke virtual method 'int android.bluetooth.BluetoothAdapter.getState()' on a null object reference
                                                                            at com.idevicesinc.sweetblue.P_AndroidBluetoothManager.getState(P_AndroidBluetoothManager.java:91)
                                                                            at com.idevicesinc.sweetblue.P_BleManager_Listeners.getBleState(P_BleManager_Listeners.java:715)
                                                                            at com.idevicesinc.sweetblue.P_BleManager_Listeners.update(P_BleManager_Listeners.java:530)
                                                                            at com.idevicesinc.sweetblue.BleManager.update(BleManager.java:3191)
                                                                            at com.idevicesinc.sweetblue.BleManager$UpdateRunnable.run(BleManager.java:3380)
                                                                            at com.idevicesinc.sweetblue.P_SweetBlueThread$SweetRunnable.run(P_SweetBlueThread.java:92)
                                                                            at com.idevicesinc.sweetblue.P_SweetBlueThread$HandlerRunner.run(P_SweetBlueThread.java:133)
                                                                            at java.lang.Thread.run(Thread.java:761)`

ryanidev, 2017-08-01:
This has been fixed in a commit that has yet to be merged to master. Is this in an app on the store?

kinber95, 2017-08-01:
Yes , my company buy the library. How I get the newest version ?

ryanidev, 2017-08-01:
It's not released yet, hence why it's not merged into master. Please email us at sweetblue@idevicesinc.com, so I can give you some additional information.

closed 2017-08-02 by kinber95