iDevicesInc/SweetBlue #351: Writing data issues on One Plus 5

linkinamine, 2017-08-09:
The sdk was tested on various devices Samsung Samsung S5 S6 S7 and Sony Xperia Z3, Huawei P8, LG G4, Nexus5 and seems to be working good for typical use cases : Scan, connect, read, write

linkinamine, 2017-08-09:
Done comment updtated so I am writing a 2o bytes key to BT device and in all other mentionned phones it worked seamlessily, in the one plus 5 I kept getting the aforementionned error.

ryanidev, 2017-08-14:
Is the characteristic you're writing to actually a WRITE_NO_RESPONSE?

linkinamine, 2017-08-17:
yes it is a  WRITE_NO_RESPONSE .

Petrulak, 2017-08-30:
Just to give a little bit more context about the issue.
The scenario is that we are connecting to a ring lock, which is mounted on a bike. We are opening it and closing it afterward.

This scenario consists of several steps, which are being done in code our code:
1. Scan & connect to the device.
2. Enable notifications, to get the change of the state.
3. Write secret token (5x20 bytes) to activate the lock.
4. Write OPT (20 bytes) which opens the lock.
5. Write OPT (20 bytes) which closes the lock.

Step `5.`  (closing the lock) sometimes takes pretty long time on `One Plus 5` or it doesn't work at all. 
It works fine with `Nexus 5`, `Nexus 5X`, `Moto G2/3`, `Samsung S5/S6/S7/S8`, `Huawei P8`, `LG G4` and `Sony Xperia Z3` so we are assuming that there is some issue with `One Plus 5`

Please see the logs for writing which took around 7 seconds.
```
08-30 12:03:24.252 5461-5461/one.noatech.rider.debug I/NoaBleSdk: writing : 54120E4F0FEBB6CC122AF52683261DDFFA069C3E
08-30 12:03:24.254 5461-5461/one.noatech.rider.debug I/P_TaskQueue: MAIN print() - Read(EXECUTING axa:5af8d57c4f83b8284cbd_94AD 00001524-e513-11e5-9260-0002a5d5c51b txn==null 181460627 ) [Write(QUEUED axa:5af8d57c4f83b8284cbd_94AD 00001525-e513-11e5-9260-0002a5d5c51b txn!=null 76819016 )]
08-30 12:03:29.496 5461-5461/one.noatech.rider.debug W/BleManager: MAIN update() - Update loop took longer to run than the current interval of 20ms
08-30 12:03:31.436 5461-10322/one.noatech.rider.debug W/BluetoothGatt: onCharacteristicRead() - Device=CC:EC:9D:3D:94:AD handle=14 Status=0
08-30 12:03:31.437 5461-5461/one.noatech.rider.debug I/P_BleDevice_Listeners: MAIN onCharacteristicRead_updateThread() - char=00001524-e513-11e5-9260-0002a5d5c51b
08-30 12:03:31.437 5461-5461/one.noatech.rider.debug I/P_BleDevice_Listeners: MAIN onCharacteristicRead_updateThread() - GATT_SUCCESS(0) 
08-30 12:03:31.438 5461-5461/one.noatech.rider.debug I/PA_Task: MAIN setState() - Read(SUCCEEDED axa:5af8d57c4f83b8284cbd_94AD 00001524-e513-11e5-9260-0002a5d5c51b txn==null 181460627 ) - 8112
```

ryanidev, 2017-09-26:
For a test, try bonding before connecting to the OnePlus 5. Some phones need to be bonded in order for them to work correctly (even if you're not using default bluetooth bonding/pairing).

dtingz, 2018-01-17:
"try bonding before connecting to the OnePlus 5" did it helped?