iDevicesInc/SweetBlue #365: Scan Timoeut Listener

Petrulak, 2017-08-25:
Hello, is there a way how to detect Scan timeout?
Thank you

ryanidev, 2017-08-25:
BleManager has a StateListener to can determine this with.

```
blemgr.setListener_State(new StateListener()
        {
            @Override
            public void onEvent(StateEvent e)
            {
                if (e.didExit(BleManagerState.SCANNING) && !mgr.isScanning())
                {
                     // Do what you want here.
                }
            }
        });
```

SweetBlue does periodic scans (unless you force an infiinte scan via the ScanOptions class when calling startScan()), so the isScanning check is to make sure that scanning is in fact stopped.

Petrulak, 2017-08-25:
Amazing, thanks a lot @ryanidev !

closed 2017-08-25 by Petrulak