iDevicesInc/SweetBlue #376: IndexOutOfBoundsException on disconnect

joebroni, 2017-09-11:
I am getting the following error in my app when trying to disconnect a device. It is currently the top crash that I am seeing in the app's crashlytics report.

`Fatal Exception: java.lang.IndexOutOfBoundsException: Index: 5, Size: 3
       at java.util.ArrayList.get(ArrayList.java:411)
       at com.idevicesinc.sweetblue.P_TaskQueue.clearQueueOf(P_TaskQueue.java:546)
       at com.idevicesinc.sweetblue.BleDevice.disconnectWithReason(BleDevice.java:5991)
       at com.idevicesinc.sweetblue.BleDevice.disconnect_private(BleDevice.java:3570)
       at com.idevicesinc.sweetblue.BleDevice.disconnect(BleDevice.java:3536)`

ryanidev, 2017-09-11:
Joe,

 FYI, if you upload your app to an app store, you're required to purchase a commercial license.

What version is in the app? That was fixed in 2.52.10.

joebroni, 2017-09-25:
Hi Ryan,

Apparently we weren't running the newest version of the code. We have already purchased a commercial license.

Thanks,
Joe

closed 2017-09-25 by joebroni