iDevicesInc/SweetBlue #384: Application crash : NLP on BluetoothLeScanner.startScan

AndZp, 2017-09-26:
I got a crash report in Fabric: 
Phone : Samsung Galaxy S7, Android 6.0
SweetBlue ver: 2_52_09

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.le.BluetoothLeScanner.startScan(java.util.List, android.bluetooth.le.ScanSettings, android.bluetooth.le.ScanCallback)' on a null object reference
       at com.idevicesinc.sweetblue.compat.L_Util.startScan(L_Util.java:207)
       at com.idevicesinc.sweetblue.compat.M_Util.startNativeScan(M_Util.java:57)
       at com.idevicesinc.sweetblue.P_AndroidBluetoothManager.startMScan(P_AndroidBluetoothManager.java:234)
       at com.idevicesinc.sweetblue.P_ScanManager.startMScan(P_ScanManager.java:667)
       at com.idevicesinc.sweetblue.P_ScanManager.startScanPostLollipop(P_ScanManager.java:606)
       at com.idevicesinc.sweetblue.P_ScanManager.startScan(P_ScanManager.java:93)
       at com.idevicesinc.sweetblue.P_ScanManager.update(P_ScanManager.java:299)
       at com.idevicesinc.sweetblue.BleManager.update(BleManager.java:3261)
       at com.idevicesinc.sweetblue.BleManager$UpdateRunnable.run(BleManager.java:3429)
       at com.idevicesinc.sweetblue.P_SweetBlueThread$SweetRunnable.run(P_SweetBlueThread.java:84)
       at com.idevicesinc.sweetblue.P_SweetBlueThread$HandlerRunner.run(P_SweetBlueThread.java:125)
       at java.lang.Thread.run(Thread.java:818)

ryanidev, 2017-09-26:
Thanks for the report, we'll get it fixed in the next release. This is most likely happening due to Bluetooth not being on.

Have you purchased a license with us? If you upload your app using SweetBlue to an app store, it is required.

AndZp, 2017-09-27:
Thank you for reply.  We know about  a license. The app is not published yet. Our team will contact you before release.

ryanidev, 2017-10-25:
Fixed in 2.52.12

closed 2017-10-25 by ryanidev