iDevicesInc/SweetBlue #387: Data is truncated on large writes

joebroni, 2017-10-03:
Were are updating OTA updates for a device and have noticed that in a recent update, there was a change so that only up to 23 bytes can be written to a device. Here is an example of data writes 

Sweet blue 2.52.01 - 2.52.11 (only writes up to 23 bytes)
SWEET_WILL_WRITE: 01 38 00 00 C7 FF 17 
SWEET_DID_WRITE:   01 38 00 00 C7 FF 17 
SWEET_WILL_WRITE: 01 32 01 00 01 CB FF 17 
SWEET_DID_WRITE:   01 32 01 00 01 CB FF 17 
SWEET_WILL_WRITE: 01 39 83 00 01 D0 00 00 40 00 20 11 68 01 00 15 9D 01 00 15 9D 01 00 08 B5 02 4B 83 F3 08 88 03 F0 9C FA 00 40 00 20 F0 B5 47 46 80 B4 72 B6 00 22 73 4B 00 21 1A 60 80 22 72 48 D2 00 06 F0 B0 FD 80 22 70 48 00 21 D2 00 06 F0 AA FD 00 23 6E 4F 98 46 3E 1C FF 33 00 25 9C 46 10 36 63 46 78 59 7A 5D 98 43 00 2A 12 D0 41 46 4C 1C 43 46 A0 46 51 1E CA B2 90 44 42 46 5B 00 54 00 F3 18 34 19 1A 78 59 78 12 18 6D D4 17 
SWEET_DID_WRITE:  01 39 83 00 01 D0 00 00 40 00 20 11 68 01 00 15 9D 01 00 15 9D 01 00 `

Sweet blue 2.51.58 (working as expected)
SWEET_WILL_WRITE: 01 38 00 00 C7 FF 17 
SWEET_DID_WRITE:   01 38 00 00 C7 FF 17 
SWEET_WILL_WRITE: 01 32 01 00 01 CB FF 17 
SWEET_DID_WRITE:   01 32 01 00 01 CB FF 17 
SWEET_WILL_WRITE: 01 39 83 00 01 D0 00 00 40 00 20 11 68 01 00 15 9D 01 00 15 9D 01 00 08 B5 02 4B 83 F3 08 88 03 F0 9C FA 00 40 00 20 F0 B5 47 46 80 B4 72 B6 00 22 73 4B 00 21 1A 60 80 22 72 48 D2 00 06 F0 B0 FD 80 22 70 48 00 21 D2 00 06 F0 AA FD 00 23 6E 4F 98 46 3E 1C FF 33 00 25 9C 46 10 36 63 46 78 59 7A 5D 98 43 00 2A 12 D0 41 46 4C 1C 43 46 A0 46 51 1E CA B2 90 44 42 46 5B 00 54 00 F3 18 34 19 1A 78 59 78 12 18 6D D4 17 
SWEET_DID_WRITE:   01 39 83 00 01 D0 00 00 40 00 20 11 68 01 00 15 9D 01 00 15 9D 01 00 08 B5 02 4B 83 F3 08 88 03 F0 9C FA 00 40 00 20 F0 B5 47 46 80 B4 72 B6 00 22 73 4B 00 21 1A 60 80 22 72 48 D2 00 06 F0 B0 FD 80 22 70 48 00 21 D2 00 06 F0 AA FD 00 23 6E 4F 98 46 3E 1C FF 33 00 25 9C 46 10 36 63 46 78 59 7A 5D 98 43 00 2A 12 D0 41 46 4C 1C 43 46 A0 46 51 1E CA B2 90 44 42 46 5B 00 54 00 F3 18 34 19 1A 78 59 78 12 18 6D D4 17

Is there a new setting that we have to set the max length?

ryanidev, 2017-10-03:
Joe,

 You mentioned that you bought a commercial license before. What company are you with? This isn't the support channel for paying customers.

joebroni, 2017-10-03:
I am with Hidrate.

ryanidev, 2017-10-03:
Ok thanks.

BTW, your license expired in the end of June, so if you want to use any newer versions on the app store, you'll have to purchase a new license (but you can continue to use your current version as long as you want).

As for your question, 23 bytes is the standard MTU size for BLE. If you want to send larger sizes than that, you'll have to call BleDevice.setMTU() with the requested size. This will negotiate with the peripheral to get it's preferred MTU size.

Write striping is enabled by default (SweetBlue will look at the current MTU size, and if you pass in a byte array larger than that, it will split them up into multiple writes). You can turn this behavior off with BleManagerConfig.autoStripeWrites = false;.

closed 2017-10-03 by ryanidev