iDevicesInc/SweetBlue #400: Error handling in enabling notifications on characteristics

suzukieng, 2017-11-02:
I noticed that SweetBlue has a peculiar way of handling an error condition when it comes to enabling notifications on characteristics.

If the descriptor of the characteristic can not be obtained (`BluetoothGattCharacteristic.getDescriptor(uuid)` returns null), SweetBlue will consider the enabling of notifications to be successful.

Source reference: https://github.com/iDevicesInc/SweetBlue/blob/d58b9b69f1fbafcdd8a6fc032afacd028c7a7a68/library/src/main/java/com/idevicesinc/sweetblue/P_Task_ToggleNotify.java#L86

Is there any particular reason for doing it this way?

I stumbled upon it by accident. We were seeing some issues in our BLE driver when turning on notifications (BLE stack callbacks not received) and discovered that adding a short delay between the `BluetoothGatt.setCharacteristicNotification(characteristic, enable)` and the `BluetoothGatt.writeDescriptor(descriptor)` calls alleviates the issue. Just like SweetBlue we also throttle BluetoothGatt invocations/callbacks by adding delays, but in this particular scenario the two calls were performed immediately after each other. I wanted to check how SweetBlue handles these two successive calls, and was surprised that there is no throttling present.

ryanidev, 2017-11-02:
From the comment in the source you linked:
```
//--- DRK > Previously we were failing the task if the descriptor came up null. It was assumed that writing the descriptor 
//---		was a requirement. It turns out that, at least sometimes, simply calling setCharacteristicNotification(true) is enough. 
```

As far as the two operations, we haven't seen any issues when enabling notifications (or had customers report any issues with it), so we never saw the need to add a delay between the two.

closed 2017-11-02 by ryanidev

suzukieng, 2017-11-02:
@ryanidev Thanks, I saw the comment, but was wondering if there was any official info from Google or in the API that would support this kind of treatment of a null descriptor.

ryanidev, 2017-11-02:
No, it's just from experience in the field. The whole reason for this library is because the API doesn't quite work as it should :).

suzukieng, 2017-11-02:
@ryanidev I know and appreciate SweetBlue's raison d'être. We're all in this together. :)