iDevicesInc/SweetBlue #405: Scan returns wrong data

mbmc, 2017-11-08:
After bonding devices, performing a scan will return duplicated entries, once of them being wrong.

Start a scan as such:
```
BleManager.get(this).startScan(discoveryEvent -> {
                Log.d("TAG", "scan: " + discoveryEvent.device().getName_normalized()
                        + " " + discoveryEvent.device().getMacAddress());
});
// runnable to stop scan after 20s
```

logcat:
```
...
11-08 15:35:16.691 10043-10043 D/TAG: scan: MyDevice 52:D1:87:3E:B9:9F
...
11-08 15:35:16.738 10043-10043 D/BluetoothAdapter: stopLeScan()
11-08 15:35:16.742 10043-10043 D/BluetoothAdapter: isLeEnabled(): ON
11-08 15:35:17.245 10043-10043 D/BluetoothAdapter: isLeEnabled(): ON
11-08 15:35:17.250 10043-10058 D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=6 mScannerId=0
11-08 15:35:22.393 10043-10043 D/TAG: scan: sml D0:39:72:BC:24:F0
11-08 15:35:22.393 10043-10043 D/TAG: scan: sml D0:39:72:B7:1D:CE
11-08 15:35:22.393 10043-10043 D/TAG: scan: pr_bt_5b9f 00:1F:FF:3F:4F:8F
11-08 15:35:22.393 10043-10043 D/TAG: scan: <no_name> 60:03:08:B2:CA:0B
11-08 15:35:22.393 10043-10043 D/TAG: scan: <no_name> 2C:B4:3A:1E:0E:4C
11-08 15:35:22.393 10043-10043 D/TAG: scan: <no_name> F4:F5:D8:BF:23:2F
11-08 15:35:22.393 10043-10043 D/TAG: scan: MyDevice 84:A1:34:6B:16:F7
11-08 15:35:24.244 10043-10043 D/BluetoothAdapter: isLeEnabled(): ON
```

Looks like the scan is stopped then restarted for a split second. The entry returned in that secondary scan is incorrect. MyDevice should only be listed once, with the address 52:D1:87:3E:B9:9F

I've performed the same test with the stock scan, only 1 entry is returned:
```
BluetoothManager bluetoothManager =
    (BluetoothManager) getSystemService(Context.BLUETOOTH_SERVICE);
bluetoothManager.getAdapter().getBluetoothLeScanner().startScan(new ScanCallback() {
    @Override public void onScanResult(int callbackType, ScanResult result) {
        super.onScanResult(callbackType, result);
        Log.d("TAG", "scan: " + result.getDevice().getName() + " " + result.getDevice().getAddress());
           }
});
```
iPhone 7 as peripheral
Nexus 5X - Android 8.0.0 as central
SweetBlue SDK 2.52.12

mbmc, 2018-04-11:
Do you guys need more info from my side to look into this issue? Thanks.

dlayneidevices, 2018-04-12:
Hi,

Have you experienced this issue when scanning for devices other than the iPhone 7?  I tried scanning out quickly, and I do not observe this issue when bonding to devices.  Have you tried the SweetBlue Toolbox (available free here:  https://play.google.com/store/apps/details?id=com.idevicesinc.sweetblue.toolbox&hl=en)?  I am curious if you observe the same behavior when using the toolbox to scan.

Please let us know if this issue is reproducible with with devices other than the iPhone 7, and if it happens with the toobox or not.

Thanks.

mbmc, 2018-04-12:
I did some more debugging (different iPhone, different version of the library, release vs jar etc.), but it turns out that if I don't have this option
```
BleManagerConfig config = new BleManagerConfig();
config.scanReportDelay = Interval.DISABLED;
```
the scan will sometimes return a stale data. I've confirmed this behavior with the sample app `ble_util`