iDevicesInc/SweetBlue #406: Bluetooth Passkey Example

barchard, 2017-11-10:
Hi, do you have any examples of pairing to a BlueTooth device with a PassKey? Thanks

ryanidev, 2017-11-14:
Just call the bond() method in BleDevice.

closed 2017-11-14 by ryanidev

barchard, 2017-11-14:
thanks. I guess I wasn't completely clear. I was attempting to call set passkey on the raw BLE Device and disable passcode confirmation. I'm trying to pair and bond to a device programmatically for an internal project. no human intervention is desired. thanks.

ryanidev, 2017-11-14:
In that case, you can get the native BluetoothDevice instance by calling BleDevice.getNative(). Then the method you want to use would be setPin (there's a hidden method call setPassKey, but looks like it's just stubbed out to do nothing).