iDevicesInc/SweetBlue #408: com.idevicesinc.sweetblue.b.b() on a null object reference

johnpaulmckean, 2017-11-14:
My team is currently having this issue come up a lot on our testing devices. There seems to be no rhyme or reason as to why. 

We are usually connected to multiple BLE devices all with the same attributes. 

Is this possibly a problem with proguard? 

Stack trace:
`Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.idevicesinc.sweetblue.b.b()' on a null object reference
       at com.idevicesinc.sweetblue.P_BleDevice_Listeners.fireUnsolicitedEvent(P_BleDevice_Listeners.java:377)
       at com.idevicesinc.sweetblue.P_BleDevice_Listeners.onReadRemoteRssi_updateThread(P_BleDevice_Listeners.java:473)
       at com.idevicesinc.sweetblue.P_BleDevice_Listeners.access$500(P_BleDevice_Listeners.java:19)
       at com.idevicesinc.sweetblue.P_BleDevice_Listeners$7.run(P_BleDevice_Listeners.java:453)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:154)
       at android.app.ActivityThread.main(ActivityThread.java:6688)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1468)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1358)`

ryanidev, 2017-11-15:
I encountered this a couple of days ago. It will be fixed in the next release. BTW, if you upload your app to an app store, you need to purchase a commercial license. Have you done so? (If so, there's a differen support path).

johnpaulmckean, 2017-11-15:
Thanks man. When is that release? (and how often do you guys release?).

My boss is working with someone on your side now for licensing.

ryanidev, 2017-11-16:
Ok, great. I'm looking to get the next release out in a week or two.

johnpaulmckean, 2017-11-17:
@ryanidev So my team is seeing this crash constantly now. Do you know the core reason why? We'd like to move forward with your product before you may come out with a release haha.

ryanidev, 2017-11-20:
2.52.13 has been released.

closed 2017-11-20 by ryanidev