iDevicesInc/SweetBlue #415: Multiple occurances of manufacturer data not supported

xmike, 2017-12-26:
if advertisment contains more than one manufacturer data field (e.g. one in the advertising packet and another in the scan repsonse), ScanEvent does not provide consistent access to the complete manufacturer data information. 

Possible fix:
```
case DATA_TYPE_MANUFACTURER_SPECIFIC_DATA:
	// The first two bytes of the manufacturer specific data are
	// manufacturer ids in little endian.
	try
	{
		mfgId = ((scanRecord[currentPos + 1] & 0xFF) << 8) + (scanRecord[currentPos] & 0xFF);
		//mfgData = extractBytes(scanRecord, currentPos + 2, dataLength - 2);
		mfgDataStream.write(extractBytes(scanRecord, currentPos + 2, dataLength - 2));
	} catch(Exception e)
	{
		Log.e(TAG, "unable to parse manufacturer data");
	}
break;
```

ryanidev, 2018-01-02:
Thanks for the report. We'll handle this in a future update at some point. You can always parse the scan record yourself (Use BleDevice.getScanRecord() to get the whole byte array that was received from the native stack.)

ryanidev, 2018-04-16:
We're looking into this now. In your particular case, does each have a unique manufacturer id? Looking at the android source code scan record processor, you will only get one or the other (rather than combining like you're suggesting). Combining the two seems out of spec to me. If you have 2 different mfg ids then that would make more sense.

xmike, 2018-04-16:
Thanks for looking into this. Acutally, both the advertisement and scan response have manufacturer data with the _same_ manufacturer ID in our case.  The use-case is that we have some manufacturer specific data in the adverting data and - because of space limitiation - further information in the scan response.  We can't put all the manufacturer data in the scan response bec we also have proprietary centrals that do only passive scan and need some information in the adversting data.  I'm aware that the android record processor does not support this. So until now, I processed the raw data myself.
iOS devices for example concatenate data from adversting and scan response, which is, I think the correct behaviour. Looking at CSSv7 [https://www.bluetooth.com/specifications/bluetooth-core-specification]() on page 9 it says that manufacturer data may appear more than once.

ryanidev, 2018-04-19:
So I have this handled now. However, it's kind of nebulous on what the "official" behavior should be. Yes, it says in the doc that it may appear more than once, but that doesn't say anything about the data being different. So, there will instead be a List<ManufacturerData> held in the BleScanInfo within each BleDevice. Then you can concat them together. I just don't want to change the current behavior for other people, things like this tend to have adverse unattended effects.

xmike, 2018-04-30:
This sounds reasonable to me. Thanks!

closed 2018-04-30 by xmike