iDevicesInc/SweetBlue #417: Sometimes BleServer Incoming listener not calling

ryadav3, 2018-01-02:
Hi,

I am using SweetBlue library for one of our app and facing some issues with BleServer Incoming listener because it not receiving any callback even if BLE device is connected.

Please help.

Thanks.

ryanidev, 2018-01-02:
Could you give more details please? When I use the BleServer, I get callbacks in the incoming listener. BTW, if you put your app on an app store, you're required to get a commercial license (please see [the SweetBlue website for more info](https://idevicesinc.com/sweetblue)).

ryadav3, 2018-01-03:
@ryanidev Here is the code snippet which we are using to add services:

`mBleServer = BleManager.get(context).getServer(getGattDatabase());`

`private GattDatabase getGattDatabase() {`
        `return new GattDatabase()`
               ` .addService(IMMEDIATE_ALERT_SERVICE_UUID)`
               ` .addCharacteristic(ALERT_LEVEL_CHARACTERISTIC_UUID)`
               ` .setProperties().write_no_response()`
               ` .setPermissions().write()`
                `.completeService();`
    `}`
It is working most of the time but sometimes incoming listener is not calling, especially when ON/OFF the Bluetooth from device settings.

ryadav3, 2018-01-04:
@ryanidev Please suggest, how we can fix this issue?