iDevicesInc/SweetBlue #422: Rediscovery fails intermittently after disconnect

mandersen, 2018-02-07:
Expected steps to disconnect are:
1. get a reference to the connected bleDevice
2. call bleDevice.disconnect()
3. wait for a BleDeviceState.DISCONNECTED event in the BleDevice.StateListener
At that point I expect the device is disconnected and should be discoverable if the app resumes scanning (or another mobile device scans).

Actual is that these steps frequently result in a state where the bleDevice is not disconnected, not discoverable by current or other mobile devices.  Intermittent, but repros in max three tries for the reduced case.

Reduced case:
https://github.com/mandersen/sweetbluedisconnect

Repros with the ble_util example source and SweetBlue Toolbox (connect, discover services, disconnect, scan with another Android/iOS device and note that the bleDevice is not discoverable).

ryanidev, 2018-02-13:
Thanks for the report, and the sample project. However, I wasn't able to reproduce what you describe. Have you tried any other bluetooth apps for reference? (nRF Connect would be the most popular). See if the same thing happens. Also, have you tried connecting to other devices? Perhaps it's something in the firmware of the peripheral you're trying to connect to?

xmike, 2018-04-16:
I can confirm this. We are seeing this also. It is particular bad at the moment with Huawei M3 tablets. We have worked around this by turing bluetooth off and back on again after some time of trying. But that is of course not always acceptable from a user point of view.  The nRF connect app does not appear to have this issue. It always correctly disconnects from the **same** device. We also do not see this happen using an iOS version of the app.

ryanidev, 2018-04-18:
Are you bonded to the peripheral when this happens?

xmike, 2018-04-19:
Yes, the peripheral is bonded in our case.

ryanidev, 2018-04-19:
Ok I was finally able to reproduce this issue. It DOES happen with nRF as well, as it seems to be an android bug (surprise, surprise!). It only seems to happen with some phones, and only when bonded to the particular device. I couldn't get it to happen more than once on an LG G4, but with a Sony Xperia, it's very easy to reproduce.
The annoying thing is there doesn't appear to be a way to detect when this happens. When polling the device state, android reports it as being disconnected.

If you unbond the device, it will then be discoverable by other devices. Something about the bond is holding the connection it seems, but not always.

With this information, please confirm you see the same behavior. I'll keep looking into it to see if there's any way for us to know when this happens in the library.

xmike, 2018-04-30:
We finally got around to test this.  In our case we can confirm that you are correct!! If not bonded disconnect seems to always work properly. With regards to nRF Connect we have seen it there also, however it appears to happen a lot less frequent using this app compared to our code. But this may have other unrelated reasons - don't know yet.

AlejandroHCruz, 2018-09-12:
Hi, did anyone find a way to go around this, maybe by calling `unbond()` when trying to connect?

ryanidev, 2018-09-12:
Yes. some phones are different. Some require that you bond, then unbond, then it's fine. Others you have to bond, then unbond, then connect, then disconnect before it's fine.

AlejandroHCruz, 2018-09-12:
I see, interesting.
I hope it helps with the issue I'm having: Trying to connect without paring to a device that was pared before and got abruptly disconnected. Then I get this error: UNBOND_REASON_AUTH_REJECTED, which is really weird because I'm not trying to pair?