iDevicesInc/SweetBlue #435: NullPointerException P_PollManager.update(P_PollManager.java:332)

BBT-SeanMacGillicuddy, 2018-04-16:
I'm investigating SweetBlue as a replacement for our current BLE stack (RxAndroidBle) and I occasionally run into this issue:
``` 
E/ExceptionLoggingHandler: exception thrown in Thread[Thread-3,5,main]
    java.lang.NullPointerException: Attempt to invoke virtual method 'void com.idevicesinc.sweetblue.P_PollManager$CallbackEntry.update(double)' on a null object reference
        at com.idevicesinc.sweetblue.P_PollManager.update(P_PollManager.java:332)
        at com.idevicesinc.sweetblue.BleDevice.update(BleDevice.java:5934)
        at com.idevicesinc.sweetblue.P_DeviceManager.update(P_DeviceManager.java:401)
        at com.idevicesinc.sweetblue.BleManager.update(BleManager.java:3470)
        at com.idevicesinc.sweetblue.BleManager$UpdateRunnable.run(BleManager.java:3646)
        at com.idevicesinc.sweetblue.P_SweetBlueThread$SweetRunnable.run(P_SweetBlueThread.java:87)
        at com.idevicesinc.sweetblue.P_SweetBlueThread$HandlerRunner.run(P_SweetBlueThread.java:128)
        at java.lang.Thread.run(Thread.java:762)
```
 as well as the array index being out of bounds sometimes:
```
E/ExceptionLoggingHandler: exception thrown in Thread[Thread-3,5,main]
    java.lang.ArrayIndexOutOfBoundsException: length=10; index=10
        at java.util.ArrayList.get(ArrayList.java:413)
        at com.idevicesinc.sweetblue.P_PollManager.update(P_PollManager.java:330)
        at com.idevicesinc.sweetblue.BleDevice.update(BleDevice.java:5934)
        at com.idevicesinc.sweetblue.P_DeviceManager.update(P_DeviceManager.java:401)
        at com.idevicesinc.sweetblue.BleManager.update(BleManager.java:3470)
        at com.idevicesinc.sweetblue.BleManager$UpdateRunnable.run(BleManager.java:3646)
        at com.idevicesinc.sweetblue.P_SweetBlueThread$SweetRunnable.run(P_SweetBlueThread.java:87)
        at com.idevicesinc.sweetblue.P_SweetBlueThread$HandlerRunner.run(P_SweetBlueThread.java:128)
        at java.lang.Thread.run(Thread.java:762)
```
```    
void update(double timeStep) {
        for(int i = 0; i < this.m_entries.size(); ++i) {
            P_PollManager.CallbackEntry ithEntry = (P_PollManager.CallbackEntry)this.m_entries.get(i);
            ithEntry.update(timeStep);
        }

    }
```

After multiple force stops and relaunches of the application (we store MAC of a connected device and reconnect on the launch of an AndroidService) I am able to sometimes observe this issue.

Here is my config:
```           
 BleManagerConfig bleManagerConfig = new BleManagerConfig();
            bleManagerConfig.loggingEnabled = BuildConfig.DEBUG;
            bleManagerConfig.runOnMainThread = false;
            bleManagerConfig.postCallbacksToMainThread = false;
            bleManagerConfig.scanReportDelay = Interval.DISABLED;
            bleManagerConfig.defaultStatePollRate = null;
            bleManagerConfig.delayBetweenTasks = Interval.millis(350);
            bleManagerConfig.autoStripeWrites = false;
            bleManagerConfig.alwaysBondOnConnect = true;
            bleManagerConfig.scanApi = BleScanApi.PRE_LOLLIPOP;
            bleManagerConfig.tryBondingWhileDisconnected = false;
            bleManagerConfig.useLeTransportForBonding = true;
            bleManagerConfig.autoPauseResumeDetection = false;
            bleManagerConfig.autoEnableNotifiesOnReconnect = false;
            bleManagerConfig.bondingFailFailsConnection = false;
            bleManagerConfig.connectFailRetryConnectingOverall = true;
            bleManagerConfig.reconnectFilter = e -> {
                                return BleNodeConfig.ReconnectFilter.Please.persist();
            };
            bleManagerConfig.bondRetryFilter = retryEvent -> {
                                return BondRetryFilter.Please.retry();
            };
```

To maintain my interface I have had to write a reactive wrapper around SweetBlue and have done so in Kotlin.

I've noticed that recently that you updated:
```
    void onCharacteristicChangedFromNativeNotify(UUID serviceUuid, UUID charUuid, byte[] value) {
        for(int i = 0; i < this.m_entries.size(); ++i) {
            P_PollManager.CallbackEntry ithEntry = (P_PollManager.CallbackEntry)this.m_entries.get(i);
            if (ithEntry != null && ithEntry.isFor(serviceUuid, charUuid) && ithEntry.usingNotify()) {
                ithEntry.onCharacteristicChangedFromNativeNotify(value);
            }
        }

    }
```
to include a null check from a similar(?) bug report, potentially same underlying issue. 

I am unfortunately testing on an S6:
```
D/BleManager$1: UPDATE(23687) run() - BOARD: universal7420
    UPDATE(23687) run() - BOOTLOADER: g920idvu3fqi5
    UPDATE(23687) run() - BRAND: samsung
    UPDATE(23687) run() - CPU_ABI: armeabi-v7a
    UPDATE(23687) run() - CPU_ABI2: armeabi
    UPDATE(23687) run() - DEVICE: zeroflte
    UPDATE(23687) run() - DISPLAY: nrd90m.g920idvu3fqi5
    UPDATE(23687) run() - FINGERPRINT: samsung/zerofltedv/zeroflte:7.0/nrd90m/g920idvu3fqi5:user/release-keys
    UPDATE(23687) run() - HARDWARE: samsungexynos7420
    UPDATE(23687) run() - HOST: swdd6903
    UPDATE(23687) run() - ID: nrd90m
    UPDATE(23687) run() - IS_DEBUGGABLE: 
    UPDATE(23687) run() - IS_EMULATOR: 
    UPDATE(23687) run() - IS_SYSTEM_SECURE: 
    UPDATE(23687) run() - MANUFACTURER: samsung
    UPDATE(23687) run() - MODEL: sm-g920i
    UPDATE(23687) run() - PERMISSIONS_REVIEW_REQUIRED: 
    UPDATE(23687) run() - PRODUCT: zerofltedv
    UPDATE(23687) run() - RADIO: g920idvu3fqi1
    UPDATE(23687) run() - SERIAL: 1115fb3c3aaa2703
    UPDATE(23687) run() - SUPPORTED_32_BIT_ABIS: 
    UPDATE(23687) run() - SUPPORTED_64_BIT_ABIS: 
    UPDATE(23687) run() - SUPPORTED_ABIS: 
    UPDATE(23687) run() - TAGS: release-keys
    UPDATE(23687) run() - TIME: 
    UPDATE(23687) run() - TYPE: user
    UPDATE(23687) run() - UNKNOWN: unknown
    UPDATE(23687) run() - USER: dpi
```
Any help would be greatly appreciated and aid the decision to switch to SweetBlue.

ryanidev, 2018-04-16:
Thanks for the report. The offending list in the pollmanager class has now been made thread-safe. It will be included in the next release (2.52.17).

closed 2018-04-16 by ryanidev

BBT-SeanMacGillicuddy, 2018-04-16:
Brilliant! Great news. Is there an eta on 2.52.17?

ryanidev, 2018-04-16:
Best guess eta would be in the 2-3 week range.

BBT-SeanMacGillicuddy, 2018-04-16:
I'll anticipate it. Thanks again, all the effort on this lib is a godsend

ryanidev, 2018-04-16:
Thanks. Just remember, if you are using it commercially, to purchase a license before uploading to an app store.

BBT-SeanMacGillicuddy, 2018-04-16:
I hope we will be in contact regarding that soon. The justification to the business is a long process.