iDevicesInc/SweetBlue #441: Reconnect issues with status 133

rkhines, 2018-05-26:
Hi Sweetblue team,

We are seeing reconnect issues on some of our beta test devices where, after loosing the connection for some reason (out of range, airplane mode, etc.), the system fails to reconnect.  The failure seems to be at least somewhat device dependant, as it is common on a Samsung Galaxy S8 (Android 8.0.0) but we do not see the same issue on a Nexus 5X (also running Android 8.0.0)

We have the ReconnectFilter configured to retry indefinitely, and in the logs we see it retry continuously, but every attempt fails almost immediately with a gatt status 133.  In one case, the device failed 171 times to reconnect before finally succeeding.  This particular reconnect failure is one we have seen before, and one of the reason we moved to using Sweetblue in hopes that it would help alleviate some of these issues.

Is there some configuration we may be missing that may help us in this case?

Should we be detecting these failures and prompting the user to reset the BleManager?

Any suggestions you can provide would be greately appreciated.

ryanidev, 2018-06-06:
Yeah, sometimes the ble stack just gets into a state where it doesn't want to work very well. This is usually indicated by a bunch of gatt status 133 errors. Typically, recycling bluetooth will help alleviate them (calling BleManager.reset()).

ryanidev, 2018-06-06:
It's already being used by SweetBlue. It is not currently configurable. This is a BLE library after all, and I dont see why you'd want to change it to anything else.

rkh14o, 2018-06-06:
Makes sense :)

Thanks.

closed 2018-07-16 by ryanidev