iDevicesInc/SweetBlue #445: Randomly crashing due to java.lang.NoClassDefFoundError

ka05, 2018-06-20:
Please help, I cant figure out what is causing this.

```
java.lang.NoClassDefFoundError: com.idevicesinc.sweetblue.BleDevice
        at com.idevicesinc.sweetblue.P_ScanManager.handleScanEntries(P_ScanManager.java:436)
        at com.idevicesinc.sweetblue.P_ScanManager.update(P_ScanManager.java:260)
        at com.idevicesinc.sweetblue.BleManager.update(BleManager.java:3391)
        at com.idevicesinc.sweetblue.BleManager$UpdateRunnable.run(BleManager.java:3560)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:158)
        at android.app.ActivityThread.main(ActivityThread.java:7224)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1230)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1120)
```

ryanidev, 2018-06-20:
What device is this on? What's the OS level?

ka05, 2018-06-20:
Samsung Galaxy S5 
Android 6.0.1

It does work most of the time. I am able to connect to BLE devices and do what i need to to do. 

This appears to happen right after accepting permissions for location services, and only seems to happen if i accept both dialog buttons rapidly. There must be some sort of race condition occurring here.

ryanidev, 2018-06-21:
Seems like something else going on. If it were thread related, I'm pretty sure it wouldn't produce a NoClassDefFound exception. Are you using proguard/obfuscation at all?

Also, what version of the library are you using? Those line numbers aren't matching up for me on the current version.

ka05, 2018-06-21:
Im not using proguard/obfuscation.
And as for the version number im honestly not sure. I cloned the repo last year sometime and copied the source to my project and used source sets, since this repo isn't available on any CDN's maven, jitpack, google, etc. 

I supposed I should probably try pulling the latest version and replacing what i currently have?

Regarding the primary Issue, I did find a database call that was executing at the same time as the BLE scanning starts and that database call for some reason was hanging the UI Thread breifly, probably causing the SweetBlue lib to have issues. I fixed that by improving the handling around the DB code and running it on a background thread so as not to hang the UI Thread, which should have been that way before. Apparently SugarORM has some issues because it shouldnt hang the UI thread fetching one result by id, that seems preposterous. In any case I havent been able to reproduce that crash yet. Im hoping that the DB issue was the indirect cause. I will post again if I see this error again. Thank you!

ryanidev, 2018-06-21:
Sounds like you're also using a real old version if you just pulled it a year ago. Also, if you're putting your app to an app store, you are required to purchase a license (unless you are open sourcing your app).

closed 2018-06-21 by ryanidev