iDevicesInc/SweetBlue #452: BleDevice.connect(StateListener, ConnectionFailListener) calls bond()

AlejandroHCruz, 2018-09-18:
Hi it seems like the method `BleDevice.connect(StateListener, ConnectionFailListener) ` is calling `bond()` under the hood at some point. I'm unsure if this happens for devices that were bonded before or for all of them, already tried catching the error and removing the cache, but that doesn't solve the issue:

```
                if (connectionFailEvent.bondFailReason() == UNBOND_REASON_AUTH_REJECTED) {
                    connectionFailEvent.device().apply {
                        unbond()
                        clearAllData()
                    }
```

Any ideas of what could be going wrong?

ryanidev, 2018-09-18:
SweetBlue remembers if you bonded to the device prior. So it will only bond() when you did in the past for that device, or the device requires it.

If you clear data on the app, you will be fine (and you may want to make sure the option tryBondingWhileDisconnected_manageOnDisk is set to false).

By the way, in case you didn't see it, V2 is no longer supported, with the exception of critical bug fixes. We've released v3, and it has a new site here [https://sweetblue.io](https://sweetblue.io)

closed 2018-09-18 by ryanidev

AlejandroHCruz, 2018-09-18:
I see, is there any way to stop that from happening? Would `tryBondingWhileDisconnected_manageOnDisk` set to false do the trick?

We have important legacy devices that don't support pairing for regular connection, only for the firmware update and then they won't work after the update! This is a critical issue for us.

Yeah, I saw that v3 is out now, but we just implemented v2 and might take some months until we do the migration. Unless it's super simple to perform?

AlejandroHCruz, 2018-09-18:
P.S. Deleting the entire app's data is a no-go in our use case. ??

Closed #452<https://github.com/iDevicesInc/SweetBlue/issues/452>.

ryanidev, 2018-09-18:
Did you purchase a license?

AlejandroHCruz, 2018-09-18:
Yes

ryanidev, 2018-09-18:
Ok, this isnt the best place for support. You should be emailing us (sweetblue@idevicesinc.com), or really, using the forum at [https://forum.sweetblue.io](https://forum.sweetblue.io).